openshwprojects/OpenBK7231T_App #405: high power consumation in standby

botioni, 2022-11-03:
in standby it is 700 mv power consumation thets to high

mariusbach, 2022-11-03:
Depends on your device, if you have a mechanical relay switch active inside, it'll need the power to pull the magnet plus maintain wifi connection.

iprak, 2022-11-04:
What device is this?

botioni, 2022-11-04:
It is a bulb in stand by. No mechanical parts

botioni, 2022-11-04:
[image]

mikimause9, 2022-11-09:
I have the same problem with the module drawing too much current. I have three Chinese modules which I wanted to reprogram. I did it to one, but it started behaving weirdly after mounting. I thought I broke something along the way, so I reflashed the second one. Sadly it started behaving the same way the first one did. I started troubleshooting them, and found out, that with this software the transformer heats up a lot. I read online, that this behavior is very likely caused by overload. I don't know why the same module with different software takes so much more power.

valeklubomir, 2022-11-09:
I am trying to analyze effectivity of low power mode, but I think based on the Application structure and some measurements, CPU stays in low power mode only fraction of processing time. ARM968E can use low power mode (Wait for interrupt) but probably interrupts are executed with high frequency, so system scheduler wakes up frequently.

valeklubomir, 2022-11-10:
I think buying cheap devices from aliexpress carries risk having deffective pieces. Based on statistics, devices which are feezing are cheap aliexpress source. On the other hand devices from local (SK) sources are stable. 7 days already.
With buy from aliexpress I always had to consider possibility of higher ratio of deffective products.
I have to try examine more devices.
Unfortunately I do not have any devices with original firmware to do real comparision of consumption.

botioni, 2022-11-10:
I will masure tomarow I have 2 defective bulbs 1 with original firmware 1
with open bk I will put them on lab transformer to se power consumetion

mikimause9, 2022-11-10:
I know, that buying cheap Chinese devices gives a risk of poor quality, but a friend of mine sold me these modules for cheap and I wanted to use them in my "Smart home" setup. As I plan on growing my smart devices family I wanted to have them in one place, as having tons of apps from every manufacturer does not sound like a good idea. I set up a simple MQTT server to have the ability to connect more lights in the future. I'm looking forward to this project and I support it from the bottom of my heart.

iprak, 2022-11-10:
I had somewhat similar experience with W600, it clocks around 110mA and I was expecting it to be lower. My chip so far has ran fine powered by USB dongle. The 110mA is also well within the specs. W600 does use a delay based main loop instead of `rtos_init_timer` based and so is functionally simpler, maybe experimenting with W600 type main loop might reveal something.

iprak, 2022-11-13:
I can confirm that our firmware does introduce some increase the power. My testing was on W600 but I think the same applies to all devices.

* Flashed with stock demo firmware which kept waiting for user input on serial - 50mA
* Flashed with demo which continuously performed 2048 rsa calculattion - still 50mA
* OpenBKT firmware - 80mA that is quite a jump.

mikimause9, 2022-11-13:
So is there a way to lower openBKT's power consumption?

iprak, 2022-11-13:
Do we know what was the power consumption of these devices prior to being flashed? I compared the power consumption of ESP plugs and they all seemed to be around 1W.

mikimause9, 2022-11-13:
I compared the power consumption of stock fw and openBKTs on two identical modules. Stock is on average at 30 mA at 3.3v with spikes at 60 mA. The openBKTs one is constantly at 70 mA with 75mA spikes.

valeklubomir, 2022-11-13:
Based on my analysis so far. Device uses CPU sleep mode less than 10%. CPU is runnung full power 90%. Idle shows is

I found out that WiFi MAC controller generates interrupt IRQ_MAC_GENERAL at rate 500 Hz.
Also PIN quick timer generates interrupt at 250 Hz. Since this timer is shared with multiple functionalities high rate is needed.
Buttons or Device LEDs do not need such high speed timer. PWM ramping on the otherside benefits fast timer.

I would suggest optimization. Split and creation of mutliple timers would be efficient. 
Beacuse when device does not have PWM driver, lower speed (50ms) for buttons would work also. 
And when device does not have buttons/LEDs or PWM, quick timer runs without usage. 

Task Scheduler activates sleep mode only when next thread activation allows more than 2 ticks of sleep. (4ms)
Since whole WiFi is hidden in librwnx I don't know why IRQ_MAC_GENERAL is used. Normaly this interrupt would be used as base tick/poll for Wifi stack and TCP-IP stack as extra timer to spare internal CPU timer, which could be used otherwise. 
There is also extra IRQ for WIFI RX/TX operation. 

Also I detected that some devices, which have Button located on P10, which is RXD1 have collisions with UART communication
Firmware was very slow and low response time. This was caused by UART receiver generating framing or overrun interrupts.
~11000 IRQ/s generated which compares to baudrate 115200. 

FreeRTOS runs system tick at 500 Hz, but I was not able to locate place where this is acually setup. 
Changing FreeRTOSConfig.h does not have any effect. And also found at least two differnet places, were ratio ms/tick is redefined.

btsimonh, 2022-11-14:
Did you instrument the SDK to count how many times we are *actually* going into low power?
I notice around flash functions, they put something starting 'peri_...' - I assume to prevent low power?
Maybe we get stuck with it unable to go low power?

mikimause9, 2022-11-17:
had anyone any luck with making these modules power efficient? I have three of these and I'd like to be able to use them. If it is impossible then... I'd be sad :(

btsimonh, 2022-11-18:
it's something we are looking into.  But seems we still have some lwip related deaths, and so that has priority over power :(.
we will get there...  I have had my device sleeping, if only for 5% of the CPU time!.

mikimause9, 2022-12-02:
Hi, Ummm ... I know there are more important things, but has anyone accomplished any success with lowering power consumption of this module?