openshwprojects/OpenBK7231T_App #406: Home Assistant Cfg not working for driver based config

botioni, 2022-11-03:
Home Assistant Cfg not working for driver based configuration and home assistant discovery also not working

mariusbach, 2022-11-03:
It's really unhelpful to lack so many details. Device, scenario, drivers, config?

Are you using MQTT at all?

botioni, 2022-11-04:
Yes im using mqtt . I meant if you are using driver config in pin asigment then it is not generating HA  config.

GravityRZ, 2022-11-04:
i think you get a popup which states that it can not find a PWM, relay or something else

when you add a relay to the config and press the discovery button it works with me
so of we remove the popup block it should work right

botioni, 2022-11-04:
Yup right if you add driver for the if off the pop-up it sould work

iprak, 2022-11-05:
Driver based RGB should work, please add more details.

botioni, 2022-11-05:
HA cfg generator for driver based not working what more details i can tel you set a driver based config with flag 4 and se if you can generate HA config

iprak, 2022-11-05:
Are you using Discovery or copying the generate configuration? Are you getting the popup mentioned here: https://github.com/openshwprojects/OpenBK7231T_App/issues/416#issuecomment-1304529251 ?

botioni, 2022-11-05:
> getting the popup mentioned here: #416 (comment)

I dont get config and I cant use auto both are not working and yes I'm
geting the pop-up

iprak, 2022-11-05:
Ok, the popup issue was recently addressed in [1.14.129](https://github.com/openshwprojects/OpenBK7231T_App/releases/tag/1.14.129)

GravityRZ, 2022-11-05:
@biotioni 
do you mean that when you go home assistant discovery the screen is empty(no code  visible)
or that you can not use discovery because you get the popup'(which is fixed in version 129)