openshwprojects/OpenBK7231T_App #408: MQTT message led_basecolor_rgb/get use wrong value format

rogere66, 2022-11-03:
When using the color picker in Home Assistant the marker jumps around randomly. This is caused by bad value format of the led_basecolor_rgb/get MQTT message.

Messages:
 cmnd/RGBCW_Bulb/led_basecolor_rgb  send #ff40700000
 RGBCW_Bulb/led_basecolor_rgb/get   returns  FF4070

To temporarily get around this I had to comment out the rgb_state_topic line in yaml config.

openshwprojects, 2022-11-03:
Are you using the old version with invalid sprintf?
Please try updating to latest one.

rogere66, 2022-11-04:
Same problem with the very latest version, 1.14.123

openshwprojects, 2022-11-04:
@btsimonh can you check this?

valeklubomir, 2022-11-04:
Parser in cmd_newLEDDriver.c is capable to decode 5 components.
Function sendColorChange() return only 3 components. 
cmd_newLEDDriver.c : lines 549-613
cmd_newLEDDriver.c : lines 206-221
Function LED_SetBaseColor cointains bug. parsing is not stopped after 5 components. 
Sending 6 or more components could overwrite memory.

openshwprojects, 2022-11-04:
@valeklubomir  nice catch with LED_SetBaseColor buffer overwrite happening after 5 components.
Apart from that, from what I understand, the led_basecolor_rgbcw should not be used and it's not used in generated templates. It's a legacy thing. We are using led_basecolor_rgb and separate temperature and dimmer topics. So sendColorChange should always send always 3 components, otherwise sendTemperatureChange is used.

valeklubomir, 2022-11-04:
I think point is that color picker in Home Assistant sends out #ff40700000 and device return FF4070 internally in the device it is handled correctly, but probably HA is missing the last 2 components and that is probably causing issues.

btsimonh, 2022-11-04:
does HA expect 5 always, just with the last two empty?
Do we know what TAS does?

valeklubomir, 2022-11-04:
I found that appended 0000 on toher place
hass.c:194
[image]

valeklubomir, 2022-11-04:
And should not template for rgbcw utilize all 5 components?

rogere66, 2022-11-04:
The problem is most likely a missing # in front of the value - HA does then not recognize the value as hex.
HA expect only 3 values. I also removed the trailing 0000 in the yaml config.

rogere66, 2022-11-04:
Looking at the code it seems like the problem is in function sendColorChange() in cmnds/cmd_newLEDDriver.c. The snprintf() need a leading # in the format string. The same applies also to function sendFinalColor().
I don't have the SDK installed yet so have not tested this.

openshwprojects, 2022-11-04:
I am using RGBCW lamp with HA all the time so something must have changed that you have those issues.
Here is my YAML:

```
  - platform: mqtt
    name: "Table E27 RGBCW (N)"
    unique_id: "tableE27rgbcw"
    rgb_command_template: "{{ '%02x%02x%02x' | format(red, green, blue)}}"
    rgb_state_topic: "obk8D38570E/led_basecolor_rgb/get"
    rgb_command_topic: "cmnd/obk8D38570E/led_basecolor_rgb"
    rgb_value_template: "{{ value[0:2]|int(base=16) }},{{ value[2:4]|int(base=16) }},{{ value[4:6]|int(base=16) }}"
    #rgb_value_template: "{{ '%02x%02x%02x'| format(red, green, blue) }}"
    command_topic: "cmnd/obk8D38570E/led_enableAll"
    state_topic: "obk8D38570E/led_enableAll/get"
    availability_topic: "obk8D38570E/connected"
    payload_on: 1
    payload_off: 0
    brightness_command_topic: "cmnd/obk8D38570E/led_dimmer"
    brightness_scale: 100
    brightness_state_topic: "obk8D38570E/led_dimmer/get"
    brightness_value_template: "{{value}}"
    color_temp_command_topic: "cmnd/obk8D38570E/led_temperature"
    color_temp_state_topic: "obk8D38570E/led_temperature/get"
    #color_temp_value_template: "{{value}}"
    retain: true
```

but I have old HA so maybe that's not the best example...

rogere66, 2022-11-04:
Thanks, it works now with the rgb_value_template starting at first char. I also removed the leading # on the rgb_command_template (and trailing 0000).
For some reason cmnd/obk8D38570E/led_basecolor_rgb is accepted with and without a leading #.
The real issue is then that the Home Assistant Cfg example code on the web UI is wrong and need an update.

rogere66, 2022-11-04:
The line
 brightness_state_topic: "obk8D38570E/led_dimmer/get"
is also needed in yaml, but missing in HA example code. Without it the HA color picker brightness slider will go to max when selecting a color (it use the received color code to calculate the brightness (unless optimistic: false is used)).

iprak, 2022-11-05:
It seems that the rgb_value_template is incorrect .. perhaps it should be using values which @openshwprojects mentioned here https://www.elektroda.com/rtvforum/topic3880540.html#20044837.

openshwprojects, 2022-11-05:
It seems that @iprak  fixed RGB value. Thanks!

closed 2022-11-07 by openshwprojects