openshwprojects/OpenBK7231T_App #409: matt pub does not change switch

mikimause9, 2022-11-04:
Hi,
I found this project after buying a smart switch based on cb2s and I'm very excited about it. I started setting it up, but I'm a little bit stuck. I have Mosquitto set up on raspberry pi to which the module can easily connect. When I try to turn it on using this command on RPI "mosquitto_pub -m 1 -t test/26/set" I get 
`Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic test/26/set
Info:MQTT:MQTT in topic test/26/set
Info:MQTT:channelSet part topic 26/set
Info:MQTT:channelSet channel 26
Info:MQTT:MQTT client in mqtt_incoming_data_cb data is 1 for ch 26
Info:GEN:CHANNEL_Set channel 26 has changed to 1 (flags 0) `
But nothing changes on the board(the led is connected to pin 26)

openshwprojects, 2022-11-04:
Hello, thank you for using OBK. It seems you are using the command incorrectly.
You should set pin 26 to Relay and set a Channel 0 for that, and use test/0/set to set it.
The MQTT command takes CHANNEL index, not a PIN index.

mikimause9, 2022-11-04:
Wow, It worked :D Thanks for the help and thanks for sharing and maintaining such an amazing project.

closed 2022-11-04 by mikimause9

openshwprojects, 2022-11-04:
No problem. Please ask me for help if anything else goes wrong.