openshwprojects/OpenBK7231T_App #411: IR - Full technical analysis

btsimonh, 2022-11-04:
IRSend - commands are received on my Roku about 1 in 3.

Instrument to measure ISR latency & jitter for 50us ISR.

Measure IR output for frequency and duty cycle.

If bad, 
Is there a better way to control TX (another ISR running in sync with PWM?  PWM ISR?

btsimonh, 2022-11-04:
We need to be able to run TX, but without RX.

Either IRRXEnable/Disable command, or run timer even if no pin (else IRSend can't work).

Also, request for disable certain protocols? (e.g. RC5 - see to of this topic: https://www.elektroda.com/rtvforum/topic3920360-30.html)

lippertmarkus, 2022-11-06:
I also found that Rx sometimes recognizes the wrong protocol. For my TV remote it showed RC6 but sending via RC6 also worked only 1 out of 3 times. I them switched to RC5 and now it works stable every time.