openshwprojects/OpenBK7231T_App #412: Edge Toggle Switching logic

valeklubomir, 2022-11-04:
I have installed standard light switched ON/OFF in my home and want to add some mini switch module into wiring.
[image]
Is any button configuration capable to toggle relay on edge of input signal change? 
Example of usage:
Switch ON -> Relay ON
Switch OFF -> RELAY OFF
Cloud toggle -> RELAY ON
Switch ON -> Relay OFF
Switch OFF -> RELAY ON
Cloud toggle -> RELAY OFF

openshwprojects, 2022-11-04:
"TglChannelOnToggle" was created for this purpose. It detects a change of state (with debouncing) and toggles target channel when it happens.

valeklubomir, 2022-11-04:
Perfect thanks.

closed 2022-11-04 by valeklubomir