openshwprojects/OpenBK7231T_App #416: can not add BP5758D driver based light to home assistant. popup is blocking this

GravityRZ, 2022-11-05:
when i try to add this light it will not let me.
when i add a fake relay to the light it will but then the light obviously is recognised wrong.

please remove the popup check and just let us broadcast mqtt config nno matter what config is used

[image]

[image]

openshwprojects, 2022-11-05:
You are correct. There is an obvious mistake in HA discovery.
I have fixed that:
https://github.com/openshwprojects/OpenBK7231T_App/commit/d6237fce61e19051675ca1e6f89bdda7f58b95f1
Can you check new build?
Btw @iprak please check fix.

iprak, 2022-11-05:
Yes that fix makes sense.

GravityRZ, 2022-11-05:
tested with build 1.14.129 and the popup is gone and discovery is working.
light however discoverd as RGBWZ instead of RGBWW by domoticz but that is another matter 
closing this one

iprak, 2022-11-05:
Guys can you check if the color temperature layout in our device or Hass is reversed?

It seems when I select a small value
[image]
it appears like this in Hass
[image]

The same when value is at the other way around
[image]
[image]

The values in Hass and device do match.

iprak, 2022-11-05:
I moved my comment as a separate issue so that this can be closed.

closed 2022-11-05 by GravityRZ