openshwprojects/OpenBK7231T_App #417: Temperature sliders in HomeAssistant and webApp don't match

iprak, 2022-11-05:
Guys can you check if the color temperature layout in our device or Hass is reversed?

It seems when I select a small value
[image]
it appears like this in Hass
[image]

The same when value is at the other way around
[image]
[image]

The values in Hass and device do match.

openshwprojects, 2022-11-05:
That's strange! Give me a minute...

iprak, 2022-11-05:
Both the UIs are correct. 

We use Mired value which is 1,000,000/Kelvin_value.  In Kelvin lower values represents warm and higher cool and that results in inverted slider for us.

HomeAssistant uses Kelvin ( https://github.com/home-assistant/architecture/discussions/564).

I found this code related to color temperature and maybe that was prior to that change or the value got moved around:
```
// the slider control in the UI emits values
//in the range from 154-500 (defined
//in homeassistant/util/color.py as HASS_COLOR_MIN and HASS_COLOR_MAX).

#define HASS_TEMPERATURE_MIN 154
#define HASS_TEMPERATURE_MAX 500
```

MQTT integration only uses Mired value (https://www.home-assistant.io/integrations/light.mqtt/). 

Not sure if it is worth accepting Kelvin values in web and then converting it into Mired.

openshwprojects, 2022-11-05:
I am not aware about this. Are you saying that they have changed the standard used by HA during last several months?
My HA version:  
[image]

iprak, 2022-11-05:
Sorry @openshwprojects  I did not see your response while I was typing.

These are the raw values which I see in hass upon selecting a temperature
[image]

[image]

```
color_temp_kelvin: 5649
color_temp: 177
```

And this is what I see in webApp and so the Mired values do match up.
[image]

I am at the latest version
[image]

I see a fix to warm/cold reversal and then switch to Kelvin in homeassistant/util/color.py earlier this year:
[image]

ReanuKeeves01, 2022-11-08:
@iprak @openshwprojects 

To add on to this issue, Home Assistant has adjusted this in the latest update, you can see it in the [release notes here](https://www.home-assistant.io/blog/2022/11/02/release-202211/#color-temperatures-in-kelvin)

iprak, 2022-11-08:
I have an idea about how to make the slide match home assistant.

closed 2022-11-11 by iprak