openshwprojects/OpenBK7231T_App #422: BK7231N Overheats? And reboots.

mikimause9, 2022-11-05:
Hi, I have a problem,
I bought a Chinese module with BK7231N and reflashed it with openshw software. It worked like it should until it stopped. Right now the module reboots every 2 minutes or so. It connects to wifi, works a little, and reboots. The logs show nothing. I have no idea what could it be. Is there a possibility, that the module overheats?

valeklubomir, 2022-11-06:
Any way to catch log before reboot? And post it here? 
In suspect of overheating, is BK7231 hot? 
Can you measure consumption of BK7231?

openshwprojects, 2022-11-06:
How the capacitors on the board look like? I had a randomly restarting device. It had a faulty electrolytic cap on the board. It was a DIY switch by moeshous

mikimause9, 2022-11-06:
I figured out, that it overheats for some reason with unofficial software. With OG one it works as it should.

closed 2022-11-06 by mikimause9