openshwprojects/OpenBK7231T_App #423: Sending RGBCW value on change to MQTT

Bacto, 2022-11-05:
Hi,

Using MQTT, when I'm changing the RGBCW value of my light, I send this: `cmnd/bulb_1/led_basecolor_rgbcw <RGBCW>`.
It works great.

[image]

The light then sends to MQTT its current color: `bulb_1/led_basecolor_rgb/get <RGB>`.
Unfortunately, I can't get back the CW value.

Could the light send its RGBCW value like this `bulb_1/led_basecolor_rgbcw/get <RGBCW>` ?

It would be very useful for HomeBridge, that publishes RGBCW values and ask for a topic to get lights values.

[image]

> Note: in HomeBridge C and W are reversed but we can switch them then in HomeBridge configuration.

Best,
Adrien

openshwprojects, 2022-11-06:
This is good idea. I can add it, it will be possible to enable it by a flag.

openshwprojects, 2022-11-06:
@Bacto added, can you check?
https://github.com/openshwprojects/OpenBK7231T_App/commit/b08342691de3be48691982f1dde10cf921e5253c

Bacto, 2022-11-06:
Wow so fast!
Seems to work! Thank you so much!

Would you considere to activate donation using GitHub sponsorship (besides Paypal)?

closed 2022-11-22 by Bacto