openshwprojects/OpenBK7231T_App #424: Future request: speedup light on when connected to AC/powered on

GravityRZ, 2022-11-05:
When a  native Tuya light bulb is turned on it switches on instantly, you can not tell it is a smart lightbulb.
With OpenBK installed it takes the light about 0.5-1 second to turn on

is there any way to speedup this time.

e.g. when switched on with a wallswitch(it connects the light to AC) it needs to turn on near instantly.

Tuya/philips hue  do it out of the box so this must be possible

botioni, 2022-11-05:
flas ota x.x.106 and it will

botioni, 2022-11-05:
futer request remove or make it configurable the startup delay

openshwprojects, 2022-11-05:
@btsimonh there is a delay that is kinda forced by LWIP library, right?

botioni, 2022-11-06:
[image]
Here we have Btn thet is acting like you push it it is changing from 0 to 1 and staying until it is pushed again   with push buton I mean if you push it changes from 0 to 1 and if you release it chges back .  You push it it is 1 you hold it it is 1 you release it it is back to 0. I hope I make mine self clear .

botioni, 2022-11-06:
start drivers beffor delay  parhaps

btsimonh, 2022-11-06:
Ok, so that startup delay is because we run all of our 'startup' bits in one hit, and some of those did things with TCP/IP.

The problem was that doing things with TCP/IP before TCP/IP was initialised was dangerous.
We should split the startup into two sections - one which does things like read the config and the existing dynamic values, and apply them, and the second part to do anything net related.

Then the startup delay can be move to between them, only delaying the things that could need TCP/IP - being careful that any drivers started don't attempt to use MQTT, etc.

I can;t help for a few days, but it's not a difficult PR to construct with care?

The reason that TCP/IP takes so long to be initialise is (I believe) that the core thread does this, and it 'calibrates' wifi/bt first, which takes > 500ms of looping and measuring TX levels?  (and @botioni - may WRITE the RF flash area???? - so RF flash being destroyed could have been because of a flash clash - our thread trying to write some dynamic data, whilst their code writes RF Area ?)

GravityRZ, 2022-11-06:
@btsimonh 
that sounds logical.
first do the local stuff 
initialize anything network related and after that do the rest.

this is probably how tuya/hue also do their things.

if network initialization takes a little bit longer is not a problem(at the moment i think it connects within 6 seconds
people who never switch their lights with an external switch do not experience problems
people who use the smartlight as a stardard bulb(using external switch) will not experience delays.

for me it is not a big deal but purely cosmetic so treat it as such

if it makes sense change the sequence and if it is to difficult or can cause more problems leave it as it is.

valeklubomir, 2022-11-06:
May be skip the startup delay force by FW and add it as command in startup cmd or autoexect.bat

- Reorder startup command: Start things which do not requirre TCP/IP like PWM output or Relay startup state first , then place startup delay and after that commands which need TCP-IP running.
- Or create extra command "WaitTCPIP" maybe just delay or active check. Place as part of command.
- More system friendly: I would maybe inspire by LINUX system, where each system service is started in seperate process (thread) and it check if all required services are online before start it-self. Similar approach split backlog into single threads commands and update each command to check if required function is active. In this case if TCP/IP is started.

btsimonh, 2022-11-12:
Try the binaries from the zip at the bottom of the latest PR against alpha.

https://github.com/openshwprojects/OpenBK7231T_App/actions/runs/3450516404

This splits the startup into 'before delay' and 'after delay', and SHOULD restore the light values early.

Note that early.bat is run before delay, and autoexec.bat is run after the delay.

Also the delay is no longer fixed, but instead waits for the device to be ready....

GravityRZ, 2022-11-12:
would that be this file
OpenBK7231T_453_merge_5b46a61f1a55.rbl

botioni, 2022-11-12:
Yes and confirming working

GravityRZ, 2022-11-12:
also tested and comnfirmed that it is working
light switchesd on instantly

did we swap WW/CW(warmwhite now on the left and cool white on the right)

GravityRZ, 2022-11-12:
downloaded 1.15.21 but this change is not in this version.

botioni, 2022-11-12:
Yes because it is pree alpha release

btsimonh, 2022-11-13:
in the 'releases' list, there is now an alpha release.  to find it you have to go to the releases list.
This now includes this PR and OTA overwrite PR.
Treat with care, as it is pre-release.

openshwprojects, 2023-02-22:
Well, lots of time passed and I think it's implemented now, so I am closing.

closed 2023-02-22 by openshwprojects