openshwprojects/OpenBK7231T_App #426: future request pusbutton config

botioni, 2022-11-05:
future request pusbutton config

openshwprojects, 2022-11-05:
Hello, what do you mean by pushbutton config?
We can already configure things like hold/repeat times, etc.

botioni, 2022-11-06:
[image]

Here we have Btn thet is acting like you push it it is changing from 0 to 1 and staying until it is pushed again with push buton I mean if you push it changes from 0 to 1 and if you release it chges back . You push it it is 1 you hold it it is 1 you release it it is back to 0. I hope I make mine self clear .

openshwprojects, 2022-11-07:
It already is in SDK - it's called "dInput", digital input.

botioni, 2022-11-07:
dinput dosent act like a button dosent have events dosent change leds nothing not god

botioni, 2022-11-07:
i need pushbutton thet trigers led relay and has events like in tasmota ther is 20 type off butons