openshwprojects/OpenBK7231T_App #428: future request with PUBLISH command

botioni, 2022-11-06:
future request with PUBLISH command to publish from this "Publishes data by MQTT. The final topic will be obk0696FB33/[Topic]/get" to this Publishes data by MQTT. The final topic will be [Topic]/get"

valeklubomir, 2022-11-06:
Can you provide some example, beacuse this description hard to understand.

botioni, 2022-11-06:
now if you bublish somthing it will publish in the same topic as you publish values ..
i want to control other device topic to control it witin mqtt broker . ak send a command to torn light on or off or change dimm value or color

valeklubomir, 2022-11-06:
What do you want to do?
Device Publishes values on "obk0696FB33/[Topic]/get" and subscribes on topic "obk0696FB33/[Topic]/set" where you can control of device value.

botioni, 2022-11-06:
control other device not this device control with one device other device

openshwprojects, 2022-11-06:
So it seems @botioni  wants to do publish on "anythingYouWrite/anything/anything2" topic... so basically on any arbitrary topic... it should be possible, but I never tried that. 

Btw, you also have a "SendGet" command, you might want to use that one.

valeklubomir, 2022-11-06:
This would require having MQTT server in device. And secondary MQTT client. 
MQTT is single ended connection type. 
Not possible that another MQTT client could listen to communication between server and other clients.
MQTT is server-client mode. Not MESH.

valeklubomir, 2022-11-06:
You have MQTT server where you can receive information from other devices. HA subscribes to MQTT server to receive all mesages. and then process them and publish neccesdsary commands. What else is needed?

valeklubomir, 2022-11-06:
I have custom Control Software, can subscribe to any topic, but I intentionaly subscribe to "<device>/+/+" which gives me filtering for single device and control loop receives messages from single device and process them. Separate control loops are then created for each device. Master control then direct events and commands between devices.

valeklubomir, 2022-11-06:
> control other device not this device control with one device other device

This gives any body any idea what is @botioni speaking of?

botioni, 2022-11-06:
If can't be done let it I will try other aproch  to accomplish what I want
and I forgot about the tasmota control I need to look in to it . And I can
try to make a group tasmotas

botioni, 2022-11-06:
add short name to commands can thet be done? to have a group off devices in the same topic somthing like 
topic/short_name+comamad/

botioni, 2022-11-06:
> So it seems @botioni wants to do publish on "anythingYouWrite/anything/anything2" topic... so basically on any arbitrary topic... it should be possible, but I never tried that.
> 
> Btw, you also have a "SendGet" command, you might want to use that one.

im using Sendget now but only one connection at the time im controling 2 devices and it will be more .

valeklubomir, 2022-11-06:
You mean global topic, which will device publish to and same time subscribe to?
One device will publish and all devices could receive. It could be possible.
This would require device to create interpreter for incomming messages and configuration on device side.

mariusbach, 2022-11-06:
I guess what he means is, for example:

1. someone sets a openBK-device to some new state, let's say toggles a relay socket to "on" (this is being published already, I know).
2. this would trigger a mqtt-publish to a user-set topic from the openBk-device.
3. the "other device" subscribes to this topic and reacts accordingly, i.e. switch another bulb on or another relay to off or on. 

I think the point here is, that either there is no higher-logic system which can send user-set mqtt-messages upon receiving a message (in this case an openBk-device message), or that the other device has non-changeable standard topics to which it reacts. Being able to publish to this would eliminate the need of an interpreting/processing system.

uff. still easier to setup an mqtt-broker on a raspberry and use node-red to process any incoming message to any outgoing message you want :)

valeklubomir, 2022-11-06:
**MQTT comminication can NOT work without MQTT-broker, on server located on local network.**
@mariusbach So as you wrote, setting mqtt broker + node-red to process MQTT message is quick and easy.  I can confirm it.
Implementing logic in devices would requirres manually upgrade each device when new version is available.

botioni, 2022-11-06:
make it local and availeble internet mqtt connect to servers :)