openshwprojects/OpenBK7231T_App #429: Future request read mqtt topic value.

botioni, 2022-11-06:
Future request read mqtt topic value.
somthing like Mqtt_read bec1_Ladvance/Led_dimmer Value 
and storre it to a variale thet you can use for dimming -10

valeklubomir, 2022-11-06:
Can you provide some example, beacuse this description hard to understand.

botioni, 2022-11-06:
subscribe to topic like bec_ladvance/Led_dimmer/get value is 50 and read it

mariusbach, 2022-11-06:
I think he wants the openBk-device to be able to receive messages for a user-set topic and react upon them. This way the openBk-device could be notified of a controlling device and react accordingly.

botioni, 2022-11-06:
yes exactly

openshwprojects, 2022-11-06:
What about using aliased commands?
1. define an alias:
 alias myUniqueName backlog led_enableAll 1; led_basecolor_rgb #FF00FF ; led_dimmer 200
2. do a MQTT publish with cmnd =  "myUniqueName" ?

openshwprojects, 2023-02-22:
This is implemented now. Even in Tasmota tyle. But also it reacts to "get".

closed 2023-02-22 by openshwprojects