openshwprojects/OpenBK7231T_App #430: weker signal after rf config restore

botioni, 2022-11-06:
i have 2x bulbs same type same place one is bricked and restored tlv part one is stoc no brick no restor 
original has Wifi RSSI: Fair (-64dBm) restored one has Wifi RSSI: Weak (-80dBm)

btsimonh, 2022-11-06:
May be good to get some examples of the TLV section from devices.
In the webapp, you can 'read rf config' -> dump.
It would be fairly simple to add 'Download RF config' in the webapp, but for the moment a copy and paste of the config into this thread?
Maybe if we had a config for a certain type of device, then restoring it would be better done by using that config with a changed MAC?  (also fairly simple to add a 'select template config' to webapp, identify MAC location in TLV, add the random or manually changed one, and write that as the patch?)

so...  pls post the TLV area from good and bad device, clearly identifying which is which?

My guess is that this config contains certain essential factory calibrations for the specific device.  I believe the T/N SDK contains a routine for creating these, but we've no experience of it... BUT..  something runs at startup (hence the delay I put in!!!).

botioni, 2022-11-06:
bk37231T Ladvance lvt original

0000: 54 4c 56 00 e0 01 00 00 00 11 11 11 5a 00 00 00 01 11 11 11 04 00 00 00 4e 61 bc 00 02 11 11 11  : TLV.à.......Z...........Na¼.....

0020: 06 00 00 00 50 8a 06 10 64 22 03 11 11 11 04 00 00 00 02 01 00 00 04 11 11 11 04 00 00 00 8e 15  : ....P..d".....................

0040: 53 01 05 11 11 11 04 00 00 00 05 00 00 00 06 11 11 11 04 00 00 00 55 00 54 03 07 11 11 11 08 00  : S.....................U.T.......

0060: 00 00 09 00 00 00 0a 00 00 00 00 22 22 22 a2 00 00 00 01 22 22 22 04 00 00 00 0f 00 00 00 02 22  : ..........."""¢...."""........."

0080: 22 22 0e 00 00 00 0f 8f 8f 8f 8f 8f 8f 91 92 93 95 96 16 96 03 22 22 22 0e 00 00 00 0e 8e 8e 8e  : "".......""".....

00a0: 8e 8e 8e 90 91 91 92 93 13 93 04 22 22 22 0e 00 00 00 0b 8b 8b 8b 8b 8b 8b 8d 8e 8e 8f 90 10 90  : .."""......

00c0: 05 22 22 22 04 00 00 00 02 00 00 00 06 22 22 22 04 00 00 00 03 00 00 00 07 22 22 22 28 00 00 00  : ."""........."""........."""(...

00e0: 94 94 94 94 94 93 93 93 93 93 92 92 92 92 92 91 91 91 91 10 90 90 90 90 90 90 90 90 90 90 90 90  : .

0100: 90 90 90 90 90 90 90 90 08 22 22 22 04 00 00 00 01 00 00 00 00 33 33 33 84 00 00 00 01 33 33 33  : .""".........333....333

0120: 04 00 00 00 08 00 00 00 02 33 33 33 04 00 00 00 0a 00 00 00 03 33 33 33 04 00 00 00 05 00 00 00  : .........333.........333........

0140: 04 33 33 33 04 00 00 00 fd 01 00 00 05 33 33 33 04 00 00 00 03 02 00 00 06 33 33 33 04 00 00 00  : .333....ý....333.........333....

0160: ff 03 00 00 07 33 33 33 04 00 00 00 ff 03 00 00 08 33 33 33 04 00 00 00 09 00 00 00 09 33 33 33  : ÿ....333....ÿ....333.........333

0180: 04 00 00 00 0a 00 00 00 0a 33 33 33 04 00 00 00 f6 01 00 00 0b 33 33 33 04 00 00 00 00 02 00 00  : .........333....ö....333........

01a0: 00 44 44 44 40 00 00 00 01 44 44 44 20 00 00 00 78 80 78 86 74 84 76 88 68 8c 68 8c 50 94 3c 9c  : .DDD@....DDD ...xxtvhhP<

01c0: 3c a2 40 a4 40 a0 40 a0 42 a0 41 9f 41 9d 3f 9d 02 44 44 44 04 00 00 00 06 02 00 00 03 44 44 44  : <¢@¤@ @ B AA?.DDD.........DDD

01e0: 04 00 00 00 df 03 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ....ß...ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0200: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0220: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0240: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0260: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0280: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

02a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

02c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

02e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0300: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0320: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0340: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0360: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0380: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

03a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

03c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

03e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0400: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0420: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

[... 162 lines cut ...]
0e60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0e80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0ea0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0ec0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0ee0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

[... cut ...]
0f80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0fa0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0fc0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0fe0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

botioni, 2022-11-06:
BK7231N Ladvance 
0000: 54 4c 56 00 f6 01 00 00 00 11 11 11 5a 00 00 00 01 11 11 11 04 00 00 00 4e 61 bc 00 02 11 11 11  : TLV.ö.......Z...........Na¼.....

0020: 06 00 00 00 38 1f 8d d9 a7 8e 03 11 11 11 04 00 00 00 f6 00 00 00 04 11 11 11 04 00 00 00 8e 15  : ....8.Ù§........ö............

0040: 53 01 05 11 11 11 04 00 00 00 4a 00 00 00 06 11 11 11 04 00 00 00 55 00 54 03 07 11 11 11 08 00  : S.........J...........U.T.......

0060: 00 00 73 00 00 00 6f 00 00 00 00 22 22 22 b8 00 00 00 01 22 22 22 04 00 00 00 0f 00 00 00 02 22  : ..s...o...."""¸...."""........."

0080: 22 22 0e 00 00 00 0b 8b 8b 8b 8b 8b 0b 8c 8d 8d 8e 8e 0e 8e 03 22 22 22 0e 00 00 00 1b 9b 9b 9b  : ""........""".....

00a0: 9b 9b 1b 9d 9e 9e 9f a0 20 a0 04 22 22 22 0e 00 00 00 17 97 97 97 97 97 17 99 9a 9a 9b 9c 1c 9c  : .   .""".......

00c0: 05 22 22 22 04 00 00 00 04 00 00 00 06 22 22 22 04 00 00 00 04 00 00 00 07 22 22 22 28 00 00 00  : ."""........."""........."""(...

00e0: 83 83 83 83 83 83 83 83 83 83 83 83 83 83 83 83 83 83 83 03 83 83 83 87 87 85 85 85 85 85 86 8a  : .

0100: 8a 8a 8a 87 87 87 87 88 08 22 22 22 04 00 00 00 01 00 00 00 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a  : ."""........ZZZZZZZZZZZZ

0120: 5a 5a 5a 5a 5a 5a 5a 5a 5a 5a 00 33 33 33 84 00 00 00 01 33 33 33 04 00 00 00 04 00 00 00 02 33  : ZZZZZZZZZZ.333....333.........3

0140: 33 33 04 00 00 00 08 00 00 00 03 33 33 33 04 00 00 00 10 00 00 00 04 33 33 33 04 00 00 00 04 08  : 33.........333.........333......

0160: 00 00 05 33 33 33 04 00 00 00 34 08 00 00 06 33 33 33 04 00 00 00 a0 0f 00 00 07 33 33 33 04 00  : ...333....4....333.... ....333..

0180: 00 00 ff 0f 00 00 08 33 33 33 04 00 00 00 73 00 00 00 09 33 33 33 04 00 00 00 6f 00 00 00 0a 33  : ..ÿ....333....s....333....o....3

01a0: 33 33 04 00 00 00 02 08 00 00 0b 33 33 33 04 00 00 00 00 08 00 00 00 44 44 44 40 00 00 00 01 44  : 33.........333.........DDD@....D

01c0: 44 44 20 00 00 00 bf 80 40 80 40 40 bf 80 b8 80 40 ff bf 80 bf 87 40 80 40 80 bf 80 b8 ff 40 ff  : DD ...¿@@@¿¸@ÿ¿¿@@¿¸ÿ@ÿ

01e0: bf 80 bf 87 40 ff 02 44 44 44 04 00 00 00 57 03 00 00 03 44 44 44 04 00 00 00 08 01 00 00 ff ff  : ¿¿@ÿ.DDD....W....DDD........ÿÿ

0200: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0220: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0240: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0260: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0280: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

02a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

02c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

02e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0300: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0320: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0340: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0360: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0380: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

03a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

03c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

03e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0400: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0420: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0440: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
[... 161 lines cut ...]
0e60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0e80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0ea0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0ec0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

[... cut ...]
0f80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0fa0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0fc0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

0fe0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

GravityRZ, 2022-11-07:
is this still needed

i had 1 device with the corrupt tlv header. with restore RF it is working again (not sure if it is 100%)
i also have the same device without the problem(flashed 137 instantly)
that one has the mac address at location

0020: 06 00 00 00 **10 5a 17 a1 c4 83** 03 11 11 11 04 00 00 00 18 01 00 00 04 11 11 11 04 00 00 00 8e 15

GravityRZ, 2022-11-07:
non restored working unit

```
0000: 54 4c 56 00 f6 01 00 00 00 11 11 11 5a 00 00 00 01 11 11 11 04 00 00 00 4e 61 bc 00 02 11 11 11  : TLV.ö.......Z...........Na¼.....

0020: 06 00 00 00 10 5a 17 a1 c4 83 03 11 11 11 04 00 00 00 18 01 00 00 04 11 11 11 04 00 00 00 8e 15  : .....Z.¡Ä.....................

0040: 53 01 05 11 11 11 04 00 00 00 00 00 00 00 06 11 11 11 04 00 00 00 55 00 54 03 07 11 11 11 08 00  : S.....................U.T.......

0060: 00 00 80 00 00 00 80 00 00 00 00 22 22 22 b8 00 00 00 01 22 22 22 04 00 00 00 00 00 00 00 02 22  : ........."""¸...."""........."

0080: 22 22 0e 00 00 00 16 95 94 93 91 90 8f 8f 8f 8e 8d 8d 0c 8c 03 22 22 22 0e 00 00 00 13 92 92 91  : "".......""".....

00a0: 90 90 8f 8f 8f 8e 8d 8d 0c 8c 04 22 22 22 0e 00 00 00 8f 8f 8f 8e 8e 8e 8d 8d 8d 8c 8a 8a 8a 8a  : .."""....

00c0: 05 22 22 22 04 00 00 00 02 00 00 00 06 22 22 22 04 00 00 00 03 00 00 00 07 22 22 22 28 00 00 00  : ."""........."""........."""(...

00e0: 94 94 94 94 94 93 93 93 93 93 92 92 92 92 92 91 91 91 91 0f 8f 8f 8f 8f 8f 8e 8e 8e 8e 8e 8d 8d  : .

0100: 8d 8d 8d 8c 8c 8c 8c 8b 08 22 22 22 04 00 00 00 00 00 00 00 09 22 22 22 0e 00 00 00 92 91 91 90  : ."""........."""....

0120: 8f 8f 8e 8d 8d 8c 8b 8b 8a 8a 00 33 33 33 84 00 00 00 01 33 33 33 04 00 00 00 08 00 00 00 02 33  : .333....333.........3

0140: 33 33 04 00 00 00 0a 00 00 00 03 33 33 33 04 00 00 00 1f 00 00 00 04 33 33 33 04 00 00 00 fc 01  : 33.........333.........333....ü.

0160: 00 00 05 33 33 33 04 00 00 00 fe 01 00 00 06 33 33 33 04 00 00 00 e8 03 00 00 07 33 33 33 04 00  : ...333....þ....333....è....333..

0180: 00 00 ff 03 00 00 08 33 33 33 04 00 00 00 0d 00 00 00 09 33 33 33 04 00 00 00 0e 00 00 00 0a 33  : ..ÿ....333.........333.........3

01a0: 33 33 04 00 00 00 08 02 00 00 0b 33 33 33 04 00 00 00 00 02 00 00 00 44 44 44 40 00 00 00 01 44  : 33.........333.........DDD@....D

01c0: 44 44 20 00 00 00 70 80 74 80 70 84 70 86 68 8c 68 8c 50 9c 40 a8 42 a8 44 ac 42 ac 43 ac 43 ae  : DD ...ptpphhP@¨B¨D¬B¬C¬C®

01e0: 43 af 44 b0 45 b0 02 44 44 44 04 00 00 00 fb 01 00 00 03 44 44 44 04 00 00 00 1b 03 00 00 ff ff  : C¯D°E°.DDD....û....DDD........ÿÿ
```

Restored unit working

```
0000: 54 4c 56 00 e0 01 00 00 00 11 11 11 5a 00 00 00 01 11 11 11 04 00 00 00 4e 61 bc 00 02 11 11 11  : TLV.à.......Z...........Na¼.....

0020: 06 00 00 00 84 e3 42 21 ac ad 03 11 11 11 04 00 00 00 1d 01 00 00 04 11 11 11 04 00 00 00 8e 15  : ....ãB!¬­.....................

0040: 53 01 05 11 11 11 04 00 00 00 1b 00 00 00 06 11 11 11 04 00 00 00 55 00 54 03 07 11 11 11 08 00  : S.....................U.T.......

0060: 00 00 10 00 00 00 10 00 00 00 00 22 22 22 a2 00 00 00 01 22 22 22 04 00 00 00 0f 00 00 00 02 22  : ..........."""¢...."""........."

0080: 22 22 0e 00 00 00 0f 8f 8f 8e 8e 8e 8d 8d 8d 8c 8b 8b 0a 8a 03 22 22 22 0e 00 00 00 11 91 91 90  : "".......""".....

00a0: 8f 8f 8e 8d 8d 8c 8b 8a 09 89 04 22 22 22 0e 00 00 00 0e 8e 8e 8d 8c 8c 8b 8a 8a 89 88 87 06 86  : .."""......

00c0: 05 22 22 22 04 00 00 00 02 00 00 00 06 22 22 22 04 00 00 00 03 00 00 00 07 22 22 22 28 00 00 00  : ."""........."""........."""(...

00e0: 90 90 90 90 90 8f 8f 8f 8f 8f 8e 8e 8e 8e 8e 8d 8d 8d 8d 0c 8c 8c 8c 8c 8c 8c 8c 8c 8c 8c 8c 8c  : .

0100: 8c 8c 8c 8c 8c 8c 8c 8c 08 22 22 22 04 00 00 00 01 00 00 00 00 33 33 33 84 00 00 00 01 33 33 33  : .""".........333....333

0120: 04 00 00 00 08 00 00 00 02 33 33 33 04 00 00 00 0a 00 00 00 03 33 33 33 04 00 00 00 09 00 00 00  : .........333.........333........

0140: 04 33 33 33 04 00 00 00 03 02 00 00 05 33 33 33 04 00 00 00 f8 01 00 00 06 33 33 33 04 00 00 00  : .333.........333....ø....333....

0160: e8 03 00 00 07 33 33 33 04 00 00 00 ff 03 00 00 08 33 33 33 04 00 00 00 10 00 00 00 09 33 33 33  : è....333....ÿ....333.........333

0180: 04 00 00 00 10 00 00 00 0a 33 33 33 04 00 00 00 00 02 00 00 0b 33 33 33 04 00 00 00 00 02 00 00  : .........333.........333........

01a0: 00 44 44 44 40 00 00 00 01 44 44 44 20 00 00 00 80 80 80 80 78 80 78 80 70 84 70 84 60 8c 58 94  : .DDD@....DDD ...xxpp`X

01c0: 56 94 56 94 54 94 54 94 55 95 54 95 55 95 54 95 02 44 44 44 04 00 00 00 ee 01 00 00 03 44 44 44  : VVTTUTUT.DDD....î....DDD

01e0: 04 00 00 00 07 00 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  : ........ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
```