openshwprojects/OpenBK7231T_App #432: Turn bulb ON on startup

ReanuKeeves01, 2022-11-06:
Relatively simple wish, when you take the power off the bulb, and then switch it back on. It's always in the OFF state. 

I'd like it to always start in the ON state when power OFF and ON again.

Furthermore i noticed that when this one boots, you'd have to manually set the temperature first to ANY value or it won't turn on at all. It would be nice to also have this value set to 'any temperature value' really on startup. If you take them from the power now, i always have to first access them through the WebGUI to 'reset' them so that everything works again in Ha/Homekit.

Many thanks in advance. 

I'm using this pin config, with a Tuya CW/WW 820lumen bulb:

[image]

valeklubomir, 2022-11-06:
Here it is:
[image]
This should do the trick.

GravityRZ, 2022-11-06:
that's odd
i have exactly the opposite no matter what flag(12) i enable or -1 at startup.

if i leave my light in the off position, take the power off and put the power back on in always goes to the ON position
i have a BP5758D driver based light

maybe something to look into

https://github.com/openshwprojects/OpenBK7231T_App/issues/401

valeklubomir, 2022-11-06:
Actually device should this way imitate original firmware which is always ON after power is back.
To allow classic operation by turning AC power OFF/ON withou smart App. And use SmartAPP to configure device state after ON.

valeklubomir, 2022-11-06:
May be new feature Flag XX to Define OFF state as startup value.

GravityRZ, 2022-11-06:
maybee i am missing the point but both flag12 as well as -1 for startup value should already do this(but dowes not work in my case)

openshwprojects, 2022-11-06:
I will have to check your report, but I will say that there is of course a way to setup light in "short startup command" by putting there a "led_enableAll 0" etc etc commands

GravityRZ, 2022-11-06:
the problem is not that the light startsup either ON or OFF
the problem is that is does not go back to the last state.

so if the light is ON, pull the plug and connect result: ON
if the light is OFF, pull the plug(or restart) and connect result: ON(should be off)

so flag 12 does not work for the powerstate, at least not with a BP5758D controlled light
start value -1 does not work with a BP5758D controlled light

i suspect it does work with PWM lights otherwise there would be way more posts about this

openshwprojects, 2022-11-06:
@GravityRZ is correct. The enableAll state was just not saved.

Please wait 15 minutes.

openshwprojects, 2022-11-06:
It should be way better now, can anyone check? Also please see new flags added.

GravityRZ, 2022-11-06:
tested it on build 1.14.137

Flag 12 now also saves powerstate(  short blink nut light stays of when it was of)

also tested flag17
when enabled and the light is off the light will turn on when changing brightness, temperature or color

so far looking good, good work

openshwprojects, 2022-11-06:
Thank you for testing, @GravityRZ , your input is very  valuable. I was not aware about this bug until today. But it's fixed now, so I am closing the issue.

closed 2022-11-06 by openshwprojects

ReanuKeeves01, 2022-11-07:
I had indeed worked out the 'startup command' solution, which worked for me. Then realized I could also simply input the startup values for the pins directly through the interface (which largely solved my issue). Upsid was that the light would always nicely come on when the power is Switched on (Wife approval rating + ) however the downside was that the brightness/temperature is always defaulted back to that setting, regardless of any former state, which is a little annoying. 

I'm gonna play around with the new update and see what works best for me, good to see there are multiple solutions to this issue.

Many thanks for the input and great to see that my request prompted another fix and ultimately a new feature. I'm gonna update my devices today and check it out immediately. 

Thanks @openshwprojects and @GravityRZ

jcam, 2025-08-25:
Looking back at this old issue, I was wanting a configuration where the light remembers the brightness and color or color temperature, but always turns "on", even if it was off before it lost power, and without any strange blinks at startup.
I have a BP5758D lamp.
Solution: 
set flag 12 -> checked
set startup command to led_enableAll 1

I thought that I needed the BP5758D_Map 2 1 0 3 4 to be in the startup command text, and using backlog BP5758D_Map 2 1 0 3 4; led_enableAll 1 worked, but it caused the light to flash on every startup in an annoying way.
Fortunately, I found that flag 12 *also* remembers the BP5758D_Map setting!

Figured I'd comment here in case anyone else finds themselves in the same situation