openshwprojects/OpenBK7231T_App #437: Future request:softstart flag

GravityRZ, 2022-11-07:
Hello,
i got feedback from the family.
they noticed i replaced the tuya light with an OpenBK version.

One thing they noticed is that the tuya(and Hue) light do not go instantly on but sort of fade.(no ramp up but just a tiny bit slower startup). Is it possible to create something like that.?
maybe and extra flag which changes the behaviour when switching the light on/off or switching between colors
I think this will also be good for the hardware and the lifetime of the bulb since powerspikes are never good.

openshwprojects, 2022-11-07:
Oh ok, good idea, wait a hour or two

openshwprojects, 2022-11-07:
Can you check latest commit?
Enable in flags.
Use commands to change the lerp speed, the default is 200 color units per second ("led_lerpSpeed 200" command).
I had a feeling that it blinked once or twice in a very strange way, there might be some bug, but now it seems to work again.
https://github.com/openshwprojects/OpenBK7231T_App/commit/4258cd2afb57812049030e5c0fc7fa057b416270
@iprak  @valeklubomir  @btsimonh  if you have some free time, please review my LED commit code, maybe I am missing something

GravityRZ, 2022-11-07:
i tested build 141
seems to work very well
no strange blinks
my first observation:  was that it was affecting colors but it turns out when you are using a color near primary (240,10,10) the color will not stay the same when you change the brightness
this is the case with and without the flaf enabled.
second observation: the time changes liniear with the brightness of the light
eg when the light is very bright it takes longer to switch off/on
i think logarithmic would be better so the effect is the same regardless of how bright the light is set.

mhat is the max setting for led_lerpSpeed 
i have set it to 500 and that gave better results when set on higher brightness

closed 2022-11-08 by GravityRZ