openshwprojects/OpenBK7231T_App #438: Automatic Tests for important OBK features - design considerations, MQTT porting?

openshwprojects, 2022-11-07:
**TLDR: Developers, what do you think about doing automatic tests on Windows for OBK in Windows OBK Build?**

OBK started as a project supporting also a Windows build. It was possible to run OBK on windows (with MQTT disabled) for some amount of time.
Unfortunatelly, due to large amount of changes, the Windows support for lost in translation.
Recently I have readded the Windows support, but only for Scripting part:
https://github.com/openshwprojects/OpenBK7231T_App/blob/main/src/win_main_scriptOnly.c
I would like to extend it and create some kind of sandbox Windows environment, where you can:
- Run the OBK on Windows just like on any other cheap
- Run automatic tests of features to early-detect any possible errors

Here are some more technical concepts and ideas that I would like to implement.
SCRIPT TESTING will look like somewhat like this (it's an early draft):
```
int g_errors = 0;
void assertExpression(const char *s, float expectedResult) {
	float res;
	res = CMD_EvaluateExpression(s,0);
	if(abs(res-expectedResult)>0.001f) {
		printf("CHECK: [%s]=%f ERROR, retuned %f\n",s,expectedResult,res);
		g_errors++;
	} else {
		printf("CHECK: [%s]=%f OK\n",s,expectedResult);
	}
}
```
```
	assertExpression("10.0",10.0f);
	assertExpression("  10.0   ",10.0f);
	assertExpression("  10.0*2   ",20.0f);
	assertExpression("  10.0*2.4   ",24.0f);
	assertExpression("  10.0+2.4   ",12.40f);
	assertExpression("$CH10*10.0",100.0f);
	assertExpression("$CH10+10.0",20.0f);
	assertExpression("10.0+$CH10",20.0f);
	assertExpression("10.0+$CH10 \r\n",20.0f);
	assertExpression("10.0+$CH10\n\r",20.0f);
	assertExpression("15.0+$CH15\n\r",30.0f);
	assertExpression("15.0/$CH15\n\r",1.0f);
	assertExpression("1.50/$CH15\n\r",0.1f);
```
**Channels testing will be more generic** and will work by plugging into the channels system and first doing commands like "led_enableAll 1", "led_baseColor_rgb FF00FF", and then checking channel values to see if they match exceptations.

Everything will happen on Windows. 

I will do that myself, except there is only single problem - the MQTT support.
So now, I have two questions.

1. I would like to hear your opinion on the idea. In general,  do you think it's worth it ,etc?
2. Is there anyone willing to help with porting MQTT library - MQTT.c file from LWIP - to windows so MQTT can work on Windows machine? How would one go with porting MQTT? At first I hoped that I can just port mqtt.c from LWIP to Windows but I am starting to worry it relies on a bit too much LWIP-specific TCP code.... I must check this out, I haven't checked yet.

I will most likely do all porting myself, I am just not sure about MQTT part

Thoughts?

openshwprojects, 2022-11-08:
It's alive:
[image]
Of course without MQTT, but it runs. Basic OBK now runs on Windows.

valeklubomir, 2022-11-11:
May be I will create similar port on LINUX.

openshwprojects, 2022-11-11:
@valeklubomir feel free to do this but let's keep it well organized.
Also, there is currently one issue with Windows port - even the LittleFS is working (I added a 'fake' flash memory access), but the issue is with Tasmota Control.
I have totally no idea, why Tasmota Control (doing HTTP communication with a device) is not working while OBK is ran on Windows, but it works on Beken.
Tasmota Control basically does:
**http://192.168.0.118/cm?cmnd=status%200**
and parses reply JSON, but in case of Windows machine, the OBK receives the packet, but Tasmota Control says "Timed out".
I suspect that I am somehow closing the socket prematurely on Windows but no sleeps and waits solves the issue.
Here is sockets code on Windows:
https://github.com/openshwprojects/OpenBK7231T_App/blob/main/src/httpserver/http_tcp_server_nonblocking.c
(I altered the server to work without any threading)

openshwprojects, 2022-11-20:
LFS REST interface testing:
[image]

Scripting testing:
[image]

openshwprojects, 2022-12-07:
Automatic tests can now spoof MQTT publishes.