openshwprojects/OpenBK7231T_App #439: Feature request: IRSend flexible delay gap / JSON syntax

heino42, 2022-11-07:
Hi all,

my projector does not accept the code sent by Tuya Smart IR / IR blaster (WB3S based).

IRRecv:
Info:IR:IR MQTT publish IR_NEC 0xF483 0x4F 0 took 3ms
Debug:IR:IR fire event took 0ms
Debug:IR:IR decode returned true, protocol 8
Debug:IR:Address=0xF483 Command=0x4F
Debug:IR:Repeat
Debug:IR: Gap **40150us**
Debug:IR: Raw-Data=0xB04FF483
Debug:IR: 32 bits
Debug:IR: LSB first
Debug:IR:IR decode returned true, protocol NEC2 (8)

IRSend:
Info:MQTT:MQTT in topic cmnd/SmartIR/IRSend
Info:IR:Delay 110ms
Info:IR:Delay **100ms**
Info:IR:IR send NEC2-F483-4F-1
 protocol 8 addr 0xF483 cmd 0x4F repeats 1

If I understand correctly, the repetition should be faster.

In order not to complicate and screw the existing syntax, is there a way to move to the Tasmota-like JSON notation - and to have the parameters optional?
E.g. 
IRsend {"Protocol":"NEC","Bits":32,"Data":2170978686,"Pulse":1,"Repeat":1,"RDelay":100,"PDelay":1000}

Cheers,
Jan

openshwprojects, 2022-11-07:
IRSend could easily check for opening { and support this syntax. The only question is - does current IRlibrary supports RDelay and PDelay? @btsimonh  ?

btsimonh, 2022-11-12:
I'm guessing that we could use the json parser ok.  But the library, not sure.
It's probably the same library used in TAS, so if someone wanted to research and report....

I too have had some issues with send;

heino42, 2022-11-19:
Cross-checkedwith a Tasmota-flashed device (firmware tasmota-ir): Different protocol identified on Beken.
Projector on:
IR_NEC2 0xF483 0x4F 1
vs
{"Protocol":"EPSON","Bits":32,"Data":"0xC12FF20D","DataLSB":"0x83F44FB0","Repeat":1}

Projector off:
IR_NEC2 0xF483 0x4E 1
vs
{"Protocol":"EPSON","Bits":32,"Data":"0xC12F728D","DataLSB":"0x83F44EB1","Repeat":1}

As Tasmota uses [IRremoteESP8266 ](https://github.com/crankyoldgit/IRremoteESP8266 )library, is it possible that not all protocols are supported yet?
The device is capable to transmit the codes (checked with original Tuya cloud firmware before flashing).

btsimonh, 2022-11-21:
that lib does seem to have a LOT more protocols, and has defines to make it not rely on Arduino.
If anyone fancies a play to try to replace what we currently use, that would be great.