openshwprojects/OpenBK7231T_App #441: Flag 12 not working as intended for LED CW/WW bulb

ReanuKeeves01, 2022-11-07:
**Describe the bug**
Flag 12, as described doesn't work as intended for my situation. Light doesn't retain the values. Both PWM pins are reset back to 0 and light is off.

[image]

State after reboot:

[image]

For these bulbs (i have many with OpenBK now); they won't respond to the  'toggle light', not untill i at least change the WW value up a little. After that it works fine again. 

For now, I managed to get it working with the CONFIGURE STARTUP:

[image]

Although this technically always switches it on to that specific state after a reboot, rather than actually remembering the state it was in (brightness, cw/ww level and on/off.) so not entirely ideal yet although i can live with this workaround.

**Firmware:**
- Version 1.15.1
- Device Tuya CW/WW filament bulb (3001686-970709 )
- Chip/model: BK7231T

ReanuKeeves01, 2022-11-08:
Let me give that a try tonight and report back.

GravityRZ, 2022-11-08:
@ReanuKeeves01 
how do you get more than 1 startup value for channel when you goto configure startup.
All i see is Channel 0 but i do not see how i can add a second or third channel so i can change that startup value

ReanuKeeves01, 2022-11-08:
[image]

this is how my STARTUP OPTIONS screen looks. 

I have this pin config:

[image]

I simply set the light to a desirable brightness/temperature value on the homescreen, note the values for channel 1/channel 2 en use them in this screen. This seems to work for me!

openshwprojects, 2022-11-08:
@ReanuKeeves01 your approach is slightly incorrect. You should be using "led_dimmer 50" and "led_temperature 500" in "short startup command", along with "backlog" . Of course, numbers here are an example.

I will add it to the FAQ.
'Example full string:
"backlog led_dimmer 50; led_temperature 300; led_enableAll 1"

ReanuKeeves01, 2022-11-08:
Thanks @openshwprojects i will give this approach a try!

openshwprojects, 2022-11-11:
@GravityRZ  @ReanuKeeves01  any news on this? Is the issue solved?

GravityRZ, 2022-11-11:
i do not have the problem with my RGBWW ligb bulb (driver based)
[image]

it seems that @ReanuKeeves01  has problems with flag 12 and his CW/WW bulb.

maybe check if that one is pwm based

ReanuKeeves01, 2022-11-11:
@GravityRZ mine is indeed a PWM based CW/WW bulb. I will rollback one of those bulbs to default settings and then try the solution with `"backlog led_dimmer 50; led_temperature 300; led_enableAll 1"`

GravityRZ, 2022-11-11:
you also might want to add 
`led_lerpSpeed 200`
gives the light a proffesional touch because it fades on/off

you also need to enable flag 18

ReanuKeeves01, 2022-11-11:
@GravityRZ gotcha, i will give that a test and report back later tonight.

ReanuKeeves01, 2022-11-12:
@GravityRZ @openshwprojects 

Firmware:

Version 1.15.1
Device Tuya CW/WW filament bulb (3001686-970709 )
Chip/model: BK7231T

For the record, that is this bulb:
[image]

PIN Config is the same as before:
[image]

**Basic situation - No flags or commands for retaining values:** 

- No startup commands
- Only Flag 7 set (for HA controls to work consistently)

In this current state, a power cycle does NOT retain values, and the light remains OFF after powering on, light comes online with the following status:

[image]

As mentioned before, in order to now toggle it on, one has to first alter the brightness and temperature values to any other value than it currently is (assuming that's zero/0) - after that toggles on fine. Power cycling again resets the bulb back to the 0,0 state and it remains off. This behavior is expected!

----

** Same settings but WITH Flag 12 set:**

Initially, the light needs to be set to anything other than the 0,0 values. For testing purposes, I've set both brightness and temperature to the max values. Set Flag 12 to active. 

After power cycling, the light comes back on (albeit with an obvious 0,5s delay) - values are retained and correct. 
Flag 12 does seem to work as intended.

----

The solution using a STARTUP COMMAND: `backlog led_dimmer 50; led_temperature 300; led_enableAll 1`
also works as intended. The obvious difference here is, this does technically not retain values but returns them to the preset values in the command. 

---

To be complete, I have also tested the startup command route WITH the `led_lerpSpeed 200` command added.
Flag 18 set, and startup command: `backlog led_dimmer 50; led_temperature 300; led_lerpSpeed 200; led_enableAll 1`

This also works as intended and the light turns on, but now with a small fade transition rather than an instant on. YAY! 
---

**CONCLUSION**

Flag 12 DOES work as it is supposed to. And therefore this issue can be closed. 

I think depending on the specific demands or wishes of the power cycle behavior you can pick which option works best for you. For now, I am going to go with the FLAG 12 option as this remains closest to what I expected, which is for the bulb to turn back on in the same state it was turned off.

My only quirk would be the seemingly small delay (~0,5-1s) when turning on the bulb. I would swear there's one OpenBK instance in my house that always turns on near INSTANT. So I will need to compare settings to see what I've done there.

For now, I'm closing the issue as resolved.

closed 2022-11-12 by ReanuKeeves01

btsimonh, 2022-11-12:
see the issue about the 0.5s delay.  I just made a PR to remove the delay for restoring values.  If you can test and comment, that would be great....

https://github.com/openshwprojects/OpenBK7231T_App/issues/424

ReanuKeeves01, 2022-11-13:
@btsimonh which release can i find this fix on? 

Interestingly, I have another non-RGB bulb (same brand, different type) in another room which seems to switch on almost instantly ALL the time without fail.

Having double checked, that one runs on version: 1.14.71 
With NO flags set, no startup command, and only values set for 'Configure Startup'.

btsimonh, 2022-11-13:
@ReanuKeeves01 - if you look in the releases LIST (not the latest), you will now find ALPHA in there too...  it's in that one.
Yep, i think 71 was before I put in the startup delay :).