openshwprojects/OpenBK7231T_App #442: Not sending Panasonic and Sony IR codes properly?

avdmr, 2022-11-08:
Great job on the IR stuff! Unfortunately the sending a Panasonic or Sony IR code doesn't seem to work on both my CBU and 
CB3S based IR blasters.

I can receive the codes:
_Debug:IR:IR decode returned true, protocol Sony (21)
Info:MQTT:Publishing val IR_Sony 0x1 0x15 0 to IRBlaster2/ir/get retain=0
Info:IR:IR MQTT publish IR_Sony 0x1 0x15 0 took 1ms
Debug:IR:IR fire event took 0ms
Debug:IR:IR decode returned true, protocol 21
Debug:IR:Address=0x1 Command=0x15
Debug:IR:Repeat
Debug:IR: Gap 24700us
Debug:IR: Raw-Data=0x95
Debug:IR: 12 bits
Debug:IR: LSB first_

_Debug:IR:IR decode returned true, protocol Panasonic (10)
Info:MQTT:Publishing val IR_Panasonic 0x1CA 0x3D 0 to IRBlaster2/ir/get retain=0
Info:IR:IR MQTT publish IR_Panasonic 0x1CA 0x3D 0 took 1ms
Debug:IR:IR fire event took 0ms
Debug:IR:IR decode returned true, protocol 10
Debug:IR:Address=0x1CA Command=0x3D
Debug:IR:Repeat
Debug:IR: Gap 71100us
Debug:IR: Raw-Data=0x813D1CA0
Debug:IR: 48 bits
Debug:IR: LSB first_

But sending the Sony code `IRsend Sony 0x1 0x15 `and reading it with my other blaster returns:
_Debug:IR:IR decode returned true, protocol UNKNOWN (0)
Info:MQTT:Publishing val IR_UNKNOWN 0x0 0 to IRBlaster/ir/get retain=0
Info:IR:IR MQTT publish IR_UNKNOWN 0x0 0 took 4ms_

The Panasonic code `IRsend Panasonic 0x1CA 0x3D` doesn't seem to get detected by the IR blasters.

openshwprojects, 2022-11-08:
Related discussion, consider posting there:
https://www.elektroda.com/rtvforum/topic3920360-60.html
@btsimonh is temporary away for a few days, but we will look into it when he comes back. He is the main contributor of IR things.

avdmr, 2022-11-29:
I was wondering if there is anything I can do to help?