openshwprojects/OpenBK7231T_App #444: Incorrect type for energy sensor class: The following entities have state class ‘measurement’ but ‘last_reset’ is missing:

otmezger, 2022-11-08:
**Describe the bug**
Hi, I uploaded the open beken firmware on a Aubee power metering plug. All is good, and Home Assistant sees the integration correctly via MQTT:

[image]

However, when I add the "energy" sensor to the Energy dashboard of home assistant, I get the following error:

```The following entities have state class ‘measurement’ but ‘last_reset’ is missing```

[image]

According to https://community.home-assistant.io/t/energy-dashboard-last-reset-missing/406733 and 
https://developers.home-assistant.io/docs/core/entity/sensor/, the sensor class should be `energy` instead of `measurement`.
Here is a quote from the forum:

> The state class for “energy” entities should be “total_increasing” 

**Firmware:**
- Version 1.14.129
- Device: Aubees power plug
- Chip/model: BK7231N
- Device config: 
[image]

otmezger, 2022-11-08:
I found this in the code, but I'm not sure how to use `total_increasing` for the energy sensor 
https://github.com/openshwprojects/OpenBK7231T_App/blob/fb8e9f129609db475c99082d172f83ac06663756/src/httpserver/hass.c#L266

openshwprojects, 2022-11-08:
Temporary solution might be adding sensors directly to YAML:
https://www.elektroda.com/rtvforum/topic3887748.html
In a meantime, @iprak , this is related to HA discovery, can you check this?

I will also try to check on my device later. My device was added manually by Yaml, it's a pretty old one.

otmezger, 2022-11-08:
Update: updated the firmware to 1.15.3 and started the MQTT discovery for Home Assistant. The same still happens.

iprak, 2022-11-08:
I will take a look.

iprak, 2022-11-09:
The documentation calls for `device_class` to be `energy`, the `state_class` can be measurement, total or total_increasing. And that is what I am seeing. `consumption_stats` is the odd ball.

[image]

otmezger, 2022-11-09:
Tahnks @iprak and @openshwprojects thats very helpful. 

@iprak I can help to test. If you let me know where in the code I can change to `total_increasing`, then I can compile the code locally and flash the device. or you produce the binary and I can flash and test. just let me know.

iprak, 2022-11-09:
Yes `total_increasing` might be more appropriate. I see the same being for another sensor generated by a Kasa switch

[image]

The device class for energycounter can be adjusted here
[image]

otmezger, 2022-11-09:
Thanks. I’ll try to rest and make a PR. 

In my home assistant, I only see energyCounter and energyCounter_last_hour. I don’t see any stats, yesterday, today and clear date. Are those not exposed. Do you know why?

otmezger, 2022-11-10:
@iprak I'm not sure we need to set the device_class. `energy` should be just fine. what we need to change is the `state_class`. I searched in the code and found a place to set it, and made a PR.I'm trying to compile now running `make`, so it is still wip.

otmezger, 2022-11-10:
Sorry guys, I was not able to build the project. [make](https://github.com/otmezger/OpenBK7231T_App/blob/main/Makefile) didn't finish. Any chance anyone can try to build? I can flash my device and see if the state_class is set correctly for energy. or any help compiling. I'm getting this error:

```
olmo@Olmos-MBP OpenBK7231T_App % make
/Library/Developer/CommandLineTools/usr/bin/make APP_NAME=OpenBK7231T TARGET_PLATFORM=bk7231t SDK_PATH=sdk/OpenBK7231T APPS_BUILD_PATH=../bk7231t_os build-BK7231
git submodule update --init --recursive --remote
cd sdk/OpenBK7231T/platforms/bk7231t/toolchain/../bk7231t_os && sh build.sh OpenBK7231T dev_20221110_214413 bk7231t
From build.sh, variables are:
APP_BIN_NAME=OpenBK7231T
APP_VERSION=dev_20221110_214413
TARGET_PLATFORM=bk7231t
USER_CMD=
Start Compile
system:Darwin
removing .o files from our folders, and OUR .o and .d files from Debug
find: ../../../apps/OpenBK7231T/src/hal/*.c: No such file or directory
find: ../../../apps/OpenBK7231T/src/libraries/*.c: No such file or directory
find: ../../../apps/OpenBK7231T/src/libraries/Arduino-IRremote-mod/*.c: No such file or directory
find: ../../../apps/OpenBK7231T/src/libraries/Arduino-IRremote-mod/src/*.c: No such file or directory
find: ../../../apps/OpenBK7231T/src/libraries/Arduino-IRremote-mod/src/private/*.c: No such file or directory
find: ../../../apps/OpenBK7231T/src/win32/*.c: No such file or directory
find: ../../../apps/OpenBK7231T/src/win32/stubs/lwip/apps/*.c: No such file or directory
not rm -rf ../tuya_common/src/*.o
not rm -rf ../tuya_common/src/driver/*.o
find: ../../../components: No such file or directory
not is ci build
echo prerequirement0
prerequirement0
make[3]: ../toolchain/windows/gcc-arm-none-eabi-4_9-2015q1/bin/arm-none-eabi-gcc: No such file or directory
===========================================================
make[3]: ../toolchain/windows/gcc-arm-none-eabi-4_9-2015q1/bin/arm-none-eabi-gcc: No such file or directory
make[3]: *** [beken378/app/ate_app.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: *** [beken378/app/app_bk.o] Error 1
Build OpenBK7231T
===========================================================
echo prerequirement1
prerequirement1
mkdir -p Debug/obj
mkdir -p Debug/bin
echo prerequirement2
prerequirement2
make[2]: *** [mp] Error 2
make[1]: *** [build-BK7231] Error 2
make: *** [OpenBK7231T] Error 2
```

iprak, 2022-11-10:
Did you just clone OpenBK7231T_App ?

otmezger, 2022-11-11:
I tried in 2 ways:
1) using the make file in the App repo. I believe it clones the sdk repos under sdk 
2) starting from the sdk (N), and putting the App repo under app/, then running b.sh. Both fail in the same spot.

iprak, 2022-11-11:
I can take a look at this. I think the energy entities just have to be `total_increasing`. Testing it right now.

otmezger, 2022-11-15:
Thanks @iprak, I just tested it with the newest firmware and it works. I can now see the energy data in my energy dashboard for Home Assistant. Great work, thank you very much

closed 2022-11-15 by otmezger