openshwprojects/OpenBK7231T_App #445: Oops. dropped firmware on the filesystem

GravityRZ, 2022-11-08:
that's what you get when doing 2 things at a time
i accidently dropped the firmware file on the filesystem (same OTA box, wrong tab)

now how can i delete that file from the filesystem

[image]

maybe make this box red instead of blue (yes i know it was my own dumb mistake)

openshwprojects, 2022-11-08:
It shows as a 0 bytes file, so basically it's not taking space. Still, I will look into it

GravityRZ, 2022-11-09:
@openshwprojects 
any news on this

if we ceate a script, can we delete it.
looking at my problem there should be a delete file option

also flashing quick ota will not get rid of this file so aparently the quick OTA save does not delete the LFS filesystem.

what does restore the fsblock do, wipe it?

GravityRZ, 2022-11-10:
delete button working :-)

was there another way to get rid of scripts/files from the LFS?
if not then that delete button comes inhandy for those files as well

closed 2022-11-10 by GravityRZ