openshwprojects/OpenBK7231T_App #451: Moes switch support

MokryPatrik, 2022-11-11:
**Describe the bug**
Hello I tried to install software on my new moes switch. Everything installed successfully, but I am unable to match the pin for button. Relay works, led works, wifi led works. I tried every pin but no luck. The button is connected to RXD1

**Firmware:**
- Version: latest
- Device: https://www.moeshouse.com/collections/light-switch/products/tuya-zigbee-smart-switch-eu-uk-push-button-wall-light-switch-no-neutral-wire-and-n-l-required-with-multi-control-assosciation?variant=39804306915409
- Chip/model: BK7231T
- Device config: https://templates.blakadder.com/athom_SW33-TAS-2US.html

openshwprojects, 2022-11-11:
I would really need to look into that. I didn't expect RXD1 to be disabled. If you need a fix for today, you could cut the trace and router another IO to the button.
But in general, that's strange, from what I see I had a device with a button on Pin10 (RXD1?) and it worked:
[image]
I will double-check on that device later...

PS: In a meantime, would you like to submit the device to teardowns database? https://openbekeniot.github.io/webapp/devicesList.html

MokryPatrik, 2022-11-11:
Thanks for fast reply,

Yes I rerouted the trace to another pin so now it works as expected.

I also have same 3 gang switch where button 2 is connected to same pin so I’ll try again with that next week. 

Here are some pictures of the board:

[image]
[image]

And of course during next week I’ll add the device to teardowns database. 

Thanks