openshwprojects/OpenBK7231T_App #454: Calibrate white balance - PWM limiter

jf-wm, 2022-11-12:
I flashed a simple LED bulb with adjustable white color with openbk, it worked flawlessly from the first try. However I would  like to be able to calibrate the white balance and/or limit the PWM range for warm white and cold white individually. The cold white LEDs in my bulb are much brighter than the warm white one, therefore the white temperature range in web app is far off from reality. I couldn't find such a feature in the configuration, maybe someone can help here?

openshwprojects, 2022-11-12:
I could add such a feature, but how would you like it to work? Can you be more specific? What kind of mapping, a linear one, how defined?

jf-wm, 2022-11-12:
Thanks for your quick answer. I think a linear mapping would be sufficient. For an easy setup it would probably be the best way to let the user define 2 points (eg. 2700k and 5500k) and use them for a linear regression.

openshwprojects, 2022-11-12:
We are internally using 0 to 1 fraction for LED temperature, which is mapped from 154 to 500 values from Home Assistant. 0 means it is 100% cold and 0% warm and 1 means 0% cold and 100% warm. 0.5 means 50%-50% etc
(_I may have swapped here cold and warm in this description_). We are not using k's internally.
I think I could implement some kind of range control, but in a different way.

jf-wm, 2022-11-12:
I just reconfirmed what you wrote with my lamp, unfortunately when adjusting the white temperature I see the following behaviour. At the left and right of the slide pwm1 or pwm2 are at 100%, however in the middle of the slide pwm1 is approx. 75% and pwm2 approx. 25%, the 50% / 50% point is is very fare of to the left. I would suspect this unlinear mixing to be the problem with my white temperate being way off from reality.
See the screenshots attached for reference. 

[image]
[image]
[image]
[image]
[image]

openshwprojects, 2023-01-12:
Please continue discussion here:
https://github.com/openshwprojects/OpenBK7231T_App/issues/599

closed 2023-01-12 by openshwprojects