openshwprojects/OpenBK7231T_App #457: BK7231N CBU module sleep

rickgartner, 2022-11-12:
I have a battery powered door sensor based on the CBU module.  I have configured it similar to the Tuya DS06 Door Sensor.  I’m able to see sensor state changes on pin 16.  I configured the startup command to include linkTuyaMCUOutputToChannel 16 val 16; setChannelType 1 ReadOnly; and it’s sending data via MQTT.  The problem is the Wi-Fi radio never sleeps and batteries last less than 24 hours.  How can I get the radio turned off and wake on a sensor state change?

valeklubomir, 2022-11-12:
At moment I am examining the SDK to somehow reduce power consumption, but without proper documentation it is very hard process. Battery powered devices will requirre totally diffrent approach. 
Shutdown every not neccessary functions. 
- Disable PIN timer(5ms). Main timer run in longer intervals
- May be deep sleep and wake on pin change
Wifi is optimized for best performance, therefore max power requirement. 
- Power save of RX, TX.
- May be reduce network communication to bare minimum. 
- May be bypass wifi access point and pair directly with another OpenBEKEN device, with mains power which would relay messages to MQTT. 

At moment whole connection to AP, DHCP and connect to MQTT takes between 8-15 seconds. Too much for battery powered device. Original firmware has pretty fast reaction <1sec. But device is not working as network device, it is paired with only one device.

This all requires deep analysis of SDK.

btsimonh, 2022-11-13:
Ref the startup time.
The 'default' 'extended app' functions (core thread) runs calibration every time.
I think this is intended to be stored in flash and just used... but we'd want to examine the calibration routine, and see exactly what it's doing, and understand what compiler flags to set to move from manual calibration to flash based calibration.
That would remove 1/2s from boot.

I suspect that now we have a delay which awaits TCP/IP and wifi stack readiness ('Split Main.... PR), we could launch wifi more quickly?

Does the original firmware actually connect to the cloud?  Or does it work with BLE?
We have BLE (at least on Beken T), but have not investigated it yet.
The battery devices I have are all BLE based, and the cheapest way of exporting the door sensor indication would be to send a BLE advert with the data (MI style).  You could then catch that in TAS ESP32 (or OpenBeken if we implemented it) for export on wifi.

iprak, 2022-11-13:
I don't know about BK7231 but W600 can be configured to enter power-saving mode to be awoken on timer0 or gpio. I suspect we can't use timer since a lot of stuff uses timers internally. But a door sensor could be awoken based on GPIO state change at which point, it can connect, send status and then go back to sleep.

btsimonh, 2022-11-15:
I have a sleepy T:
```
Info:MAIN:Time 239, idle 281511/s, sleep_ticks 19105/761436(0)(0,0,0), free 90384, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 5/38
Info:MAIN:nosleep: 166300, 0, 0, 8980, 26992, 38164, 0
Info:MAIN:Time 240, idle 261529/s, sleep_ticks 19164/762884(0)(0,0,0), free 90384, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 5/38
Info:MAIN:nosleep: 167116, 0, 0, 8980, 27125, 38316, 0
Info:GEN:dhcp=0 ip=192.168.1.176 gate=192.168.1.1 mask=255.255.255.0 mac=7c:f6:66:0b:3f:e8
Info:GEN:sta: 1, softap: 0, b/g/n
```

of 762884 sleep ticks requested by rtos, it actually slept for 19164 ticks.

no current measurements yet, but increased the pin timer to 20ms, the expected time before wake to 3 ticks, enabled sleep.
More later.... (lot's of diagnostic changes to SDK, I need to decide what was important....)

bernd-wechner, 2022-11-18:
Following with interest as I have same. Just invested in Tuya Door/Window sensors and they have a CBU IC inside, which is based on the BK7231N chip. The data sheet is here:

https://developer.tuya.com/en/docs/iot/cbu-module-datasheet?id=Ka07pykl5dk4u

The CBU IC supports WiFi and BLE, mine are Wifi and so I imagine that's Tuya's way of saying the firmware on mine uses Wifi not BLE.

I'd be curious how to experiment with flashing. I have a CP2102:

https://www.aliexpress.com/item/1859102520.html

and wonder how to connect it. My reading thus far would be to connect (CP2102 to CBU): 

RX to RX1 (there's an RX2 as well)
TX to TX1 (there's a TX2 as well)
GND to GND
3V3 to 3V3

and that's all that's needed. But admit to naivite here and the CP210 has also 5V line out (seemingly not needed)here, and a DTR (the purpose of which is unclear to me).

I'd be very curious if on reflashing then this battery life issue is resolved. Indeed, I can imagine these door sensors are very passive waking up to transmit a state change notification and then sleeping again to get the cited battery life and no-one wants to be replacing batteries more often than needed. So I'd be reluctant to reflash in that case.

I did get me one spare sensor to play around with and experiment on though, and would happily sacrifice it to test a successful reflash.

bernd-wechner, 2023-01-01:
> Also to know how long a state change takes to get from the time of door open to the time the Tuya app knows about it.
> This would give us some performance 'targets' to aspire to.

I have in the interim registered with the Tuya Cloud and can use its API using [tinytuya](https://pypi.org/project/tinytuya/).

From that, I can read on-line and off-line state changes (the device clearly sends a message just after connecting the WiFi and just before closing the Wifi connection.

Provisional readings I have are from 594 door openings, and see a rather broad range of up times, from 5.7 seconds to 432 seconds with a mean of 142 seconds (2 minutes). Fascinating. This may relate to the door open hold time. Around the 2 minutes mark strikes me as likely for the door to be help open (and the switch thus in an open state) and it may have some kind of internal logic whereby it holds the line up until the earlier of a) the switch closes again or b) some time limit (over 2.5 mins it seems).

Of course, as ever, a graph speaks volumes and here's a histogram chronologically of the up times:

[image]

Which lends some credence to the hypothesis. It seems it may have a 240s (2 min) uptime limit going down if the door sensor is closed before then, and there are some outliers which might be because:

1. drop signal (I am pairing up and down signals, and need to look at how I handle up down down say or up up down etc).
2. A timer reset on door close and open (I can drill down into this) by which I mean if inside of 2 mins, someone closes the door and then opens it again inside a grace period it does not god down then up but stays up.

All hypothesising and using signals recorded on the Tuya cloud, downloaded using their API. Oddly too, I am in Tasmania bought my sensors on eBay from China, and they are tied to the European data centre (Tuya have data centres across the world including Asia and US)

> It would be good to measure the battery current on a Tuya firmware, during no-action condition, and during a state change.

That would require me to set up a little test bench I guess. With an ammeter between the batteries and board, and watching the state (but easiest way for me to watch the state is on the Tuya Cloud - as the phone app doesn't convey that signal) or going one better and learning how to read the signals from the chip pinouts - but that is whole extra league of setup, learning and complexity. Doable, but I am madly busy and distracted and lack a good work bench for electronics at present (though I have a pile of kit for it).

> But to use Deep Sleep, we'd need to wake up on GPIO change, and then work out what we need to do to keep wifi alive or restore wifi quickly... it's a little work to ensure we're waking on configured input pins, etc.
> It's possible that we could wake every XXms to keep wifi alive, or it may be that we need to restore wifi every wake - currently we have no clue, needs investigation.

Yep, sounds hairy. I don't suppose it's easy to dump the firmware from a shop switch and reverse engineer it some. My days of reverse engineering binary are a long way in the past and it was never cheap (I'd disassemble it on the fly in a debugger an step the code - not least because you could never disassemble the stuff confidently without running it - and so would need a chip runtime environment simulator methinks because the instruction boundaries were not easy to divine up front - i.e. I'd disassemble a chunk of code, and step through it only to find that a JMP instruction or such landed me in the middle of an earlier disassembled instruction, i.e had to disassemble anew from that point - to be sure, one possible reason was that my main application here was stripping copy protection from games and they took some efforts already then to obfuscate their binaries so it might be that simple firmware plays more nicely and disassembles cleanly)

All good fun, just need the time and the workbench tee hee ... (both of which I'm hard up for in recent years - well since having children and exploding my interest bases).

bernd-wechner, 2023-01-17:
OK, I have now collected uptime statistics for a while. The sensors record an online and offline event which they send to the Tuya cloud servers. I poll those using [TinyTuya](https://github.com/jasonacox/tinytuya) and here's a summary of a months worth of events now:

[image]

Those bins are autoselected by numpy and I could bin it whatever way. It seems that a bin between 200-250 seconds would capture the mode, but that's visible in the histogram anyhow in 204-237 bin numpy home in on.