openshwprojects/OpenBK7231T_App #459: Interesting temperature/humidity monitor, possible to support?

TurkeyMan, 2022-11-13:
I have an interesting thermometer/humidity metering device, I'm wondering if it's possible to support:
[image]

It's very hack-able, just 2 screws and it opens right up, nice accessible board, very easy to break-out:
[image]
[image]

It has the most user-friendly display connector I've ever seen! It just pressure-contacts a soft rubberised edge connector which seems to be conductive.

CB3S chip; it's not mentioned on the main `README.md` document, but I see CB2S is BK7231N, so I assume it's also the same chip?
2 extra chips... what are they? I guess the big bottom one is a display controller... not really easy to see how BK7231 communicates with the other chips.

This device has a screen. I guess flashing OpenBK might be possible to read the sensors, but I guess there's no way the display will ever be supported? It probably has very device specific software for the display? Is there any way this can work in the future?
I wonder if the lower chip operates independently and just has some registers to accept values to display from BK7231, or if BK7231 must drive the display actively?

openshwprojects, 2022-11-13:
It's TuyaMCU. It is most likely already supported, but if not, would you like to help us and create a TuyaMCU packets dump?
It's very easy. TuyaMCU drives display, not BK. TuyaMCU also reads sensors. WiFI module just receives them on UART

TurkeyMan, 2022-11-13:
Oh cool. I can help with this tomorrow. What is your timezone (I'm +10)?
If we don't sync well, you can leave me detailed instructions how to collect the relevant information?

openshwprojects, 2022-11-13:
Get USB to UART converter. Connect RX of converter to RX pin of CB3S to see what it receives from TuyaMCU. Launch RealTerrm. Go to capture:
[image]
enable capture to file, play around or wait, repower device, etc etc
Get file like this:
[image]
Then I will use my analyzer:
[image]
And we will see what kind of variables are transferred.

Then repeat the packet capture, but connect USB to UART RX this time to TX pin of CB3S, to see what CB3S sends to TuyaMCU.

Do many various samples of captures.

Hmm I'm in Poland

TurkeyMan, 2022-11-13:
Okay, seems straight forward enough. I'll have a go tomorrow.
I'm can't imagine what the CB3S might want to send to the TuyaMCU... it's a sensor-only device. No switches or anything.
Does it transmit startup code or anything? Should I power-off/on this thing while monitoring?

TurkeyMan, 2022-11-13:
Oh yeah; it's battery powered... I read somewhere that OpenBK drains the battery super fast because the wifi doesn't sleep properly. Is that likely to be a problem here?

TurkeyMan, 2022-11-14:
What is the proper baud rate for TX/RX1 comms to TuyaMCU? At 9600 I get some junk... at 115200 it gets nothing.

TurkeyMan, 2022-11-14:
Okay, here's what I got:
```
Expect these values in the reports:
  Temperature is ~28.8 degrees C
  Humidity is ~50%
  Powered by 3.3v, which reports 0% battery level (normally 3x AAA)

RX pin:

[boot up]
55AA000100247B2270223A227971697162616C6474723069376D7275222C2276223A22312E312E36227D0755AA000200000155AA000200000155AA0005000509040001001755AA000500080A02000400000186A355AA000500080B020004000000001D55AA000500080C0200040000003C5A55AA000500080D020004000000143355AA0005000811020004000000012455AA0005000812020004000000012555AA0005000813020004000000062B55AA0005000814020004000000062C55AA00050008010200040000011D3155AA0005000802020004000000324655AA000600000555AA000500050E040001021E55AA000500050F040001021F55AA0005000804020004000000001655AA0005000503040001001100
[short wait]
55AA000100247B2270223A227971697162616C6474723069376D7275222C2276223A22312E312E36227D0755AA000200000155AA000200000155AA000200000155AA00100001001055AA00050008010200040000011D3155AA0005000802020004000000324655AA000600000500
[short wait]
55AA000100247B2270223A227971697162616C6474723069376D7275222C2276223A22312E312E36227D0755AA000200000155AA000200000155AA000200000155AA00100001001055AA000600000555AA000500050E040001021E55AA00050008010200040000011D3155AA0005000802020004000000324600
[press button]
55AA000100247B2270223A227971697162616C6474723069376D7275222C2276223A22312E312E36227D0755AA000200000155AA000200000155AA00100001001055AA000A00000955AA000600000555AA000A00000955AA000600000555AA000A00000955AA000A00000955AA000A00000955AA000A00000955AA000A00000900
[short wait]
55AA000100247B2270223A227971697162616C6474723069376D7275222C2276223A22312E312E36227D07
[short wait]
55AA000200000155AA000200000155AA000200000155AA00100001001055AA000600000555AA0005000801020004000001203455AA00050008020200040000003246
[short wait]
55AA0005000801020004000001203455AA00050008020200040000003145

TX pin:

[boot up]
F01838383818383838183818383838181818183838383818181838383838181838383838381818181818181818181818381838383838383838383838181838383818183838383838383818181818383838381838383818181818181818383818181838383818183838383818183818181818381818183838181838381818183818183838381818381818383818183838181818181818381818383818381818383838181838181838181838181818183818183838181838181818183818181818381818181838183818381838183838183818181838181838181818381838181818181818381818181838181818381818181818381818381818381838181838381818383838181818181818183818181818381838183818381818183818181818381838183818181818381838381818181818381838181818181838181838183818183818381818381818381818181818381818181818381818181838183818381838183818183818381818181818183818183818181818181818183818381818381838181818383818181838183818181818181838181838181838183818381838181818181818183818181818181838381818181818183818383818381838181818383818183818181818181818380055AA000100000055AA00020001020455AA00020001030555AA00020001040655AA00050001000555AA00050001000555AA00050001000555AA00050001000555AA00050001000555AA00050001000555AA00050001000555AA00050001000555AA00050001000555AA00050001000555AA00050001000555AA0006000801160B0E0C2E01017955AA00050001000555AA00050001000555AA00050001000555AA0005000100050055AA000100000055AA00020001020455AA00020001030555AA00020001040655AA0010000201001255AA00050001000555AA00050001000555AA0006000801160B0E0C2E11018900
[short wait]
0055AA000100000055AA00020001020455AA00020001030555AA00020001040655AA0010000201001255AA0006000801160B0E0C2F07018055AA00050001000555AA00050001000555AA00050001000500
[short wait]
0055AA000100000055AA00020001020455AA00020001030555AA00020001040655AA0010000201001255AA0006000801160B0E0C3009018355AA00050001000555AA00050001000500
[press button]
0055AA000100000055AA00020001020455AA00020001030555AA00020001040655AA0010000201001255AA00050001000555AA00050001000555AA0006000801160B0E0C302D01A755AA00050001000555AA00050001000555AA000A0001000A55AA000A0001010B000055AA000100000055AA00020001020455AA00020001030555AA00020001040655AA0010000201001255AA0006000801160B0E0C3109018455AA00050001000555AA00050001000500
[change 1 settings]
0055AA000100000055AA00020001030555AA00020001040655AA0010000201001255AA0006000801160B0E0C3208018455AA00050001000555AA00050001000500
[change lots of settings & wait]
0055AA000100000055AA00020001020455AA00020001030555AA00020001040655AA0010000A01011202000400000078AB55AA00050001000555AA00050001000555AA00050001000555AA0006000801160B0E0C330E018B55AA00050001000555AA00050001000555AA00050001000555AA00050001000555AA00050001000555AA00050001000555AA00050001000555AA00050001000555AA00050001000555AA00050001000555AA00050001000555AA00050001000555AA00050001000500
```

TurkeyMan, 2022-11-14:
I'm interested to see the parsed version of events...

openshwprojects, 2022-11-20:
Thread continues at: https://www.elektroda.com/rtvforum/topic3933836.html

closed 2022-11-20 by openshwprojects