openshwprojects/OpenBK7231T_App #460: ESP support

abvgdeyoj, 2022-11-13:
Probably least important at the moment, but still a good idea IMHO.

mikimause9, 2022-11-13:
I think that for esp8266 tasmota is a much better choice as it is optimized for it. This project is created mainly for microcontrollers that are not supported by tasmota.

btsimonh, 2022-11-13:
It has to be said the TAS is very comprehensive and complete.
But implementing OBK on ESP would actually not be too hard; well at least for ESP32 where we have flash.
ESP8266 1M may be a little tight ... :).

iprak, 2022-11-13:
I agree with @mikimause9 on this. Existing choices are much better for ESP both in terms of stability and functionality. I use these in addition to Tasmato; the 2nd one is developed by one of the HomeAssistant authors - [Espurna](https://github.com/xoseperez/espurna), [ESPHome](https://esphome.io/)

abvgdeyoj, 2022-11-13:
Tasmota might be better optimized for ESP8266/ESP8285 and ESP32, but when you have multiple different modules, it might be preferable for them to have the same codebase, same interface and to a reasonable extent, behavior.
I know that this is absolutely not a priority, given that there are much better alternatives on ESP*, but in time, OpenBK7231T_App hopefully will mature (already outgrown its name), possibly becoming superior to [Tasmota](https://tasmota.github.io)/[ESPurna](https://github.com/xoseperez/espurna)/[ESPHome](https://esphome.io)/[ESPEasy](https://github.com/letscontrolit/ESPEasy).