openshwprojects/OpenBK7231T_App #461: DIN rain energy meter, possible to support?

TurkeyMan, 2022-11-13:
Hi again, I have another interesting device. It's a DIN rail energy meter, and it's got a few more tricks than others I've seen.
[image]

This one has a pixel display showing all the monitor information.

It is extremely easy to open, just 2 screws, no clips, it comes apart easily.
[image]

The screen is a daughter board that lifts off some mounting pins, exposing the processor:
[image]

It's another CB3S device. It's got some extra chips on the display board. They don't look the same as the TuyaMCU chips I found in another device.

It looks like the only pins connected are P6, P9, and RX/TX, and looking at the PCB, I don't think P6/P9 go anywhere; they're just soldered to hold the thing together. So, VCC, GND, RX, TX... TuyaMCU again? Just different looking/different version chips?

openshwprojects, 2022-11-13:
Hello, can you post more information in the dedicated discussion topic?
https://www.elektroda.com/rtvforum/topic3928897.html

TurkeyMan, 2022-11-13:
You want me to post these issues as threads there instead?
This device doesn't look quite the same as the device you linked me to.

openshwprojects, 2022-11-13:
Well, the TuyaMCU packet dump would be great, same as with LCD device.

I have a working OBK device simulator on Windows machine. I can simulate your device and write support if you capture UART data and provide me the UART capture in hex format.

Hmmmm if that's a separate device then separate Elektroda thread.

Basically, we need device information for:
https://openbekeniot.github.io/webapp/devicesList.html
Posting teardown on elektroda consists posting device information, photos, teardown photos, maybe packaging photos, maybe RX/TX packets dump, etc. Optional information where the device was bought, etc

and Elektroda is our big electronics forum but still independent from 3rd parties with reliable hosting a long history (20 years of online presence)