openshwprojects/OpenBK7231T_App #462: RTL8710CF supported?

TurkeyMan, 2022-11-13:
I have another smart-plug device here:
[image]

This one seems to have an RTL8710CF in it:
[image]

The device is very easy top open; 4 screws, no clips, but the CPU board doesn't lift out easily, and it's difficult to access:
[image]

I can't see any reference to RTL chips in the main project, but I see a lengthy discussion here: https://www.elektroda.com/rtvforum/topic3815120.html
What's the state of this?

TurkeyMan, 2022-11-13:
I just found LibreTuya, which seems to support RTL chips... what is the relationship between these projects?

openshwprojects, 2022-11-13:
@TurkeyMan  LibreTuya is basically a work of a copycat who has duplicated my work, took my SDK when I started RTL support few months ago, etc.

The story is, basically:
1. I was asked to do RTL support for OBK by a certain discord members. They offered me a donation of bulbs, I posted about that on Elektroda
2. I figured out the flashing, the SDK, the compiling
3. I was asked about that by some Kuba2k2 user on forums, he didn't know how to flash RTL, it seemed like he wanted to help with OpenBeken
4. I explained him step in step how to flash RTL and showed him the SDK
5. He asked me who offered me the donation of the bulbs, so I showed him the mentioned Discord server
6. My RTL got fried so I got the delay
7. Suprisingly then, Kuba2k2 used my help to start a new project, instead of helping with OBK
8. And then, he came straight to that Discord server, with which members I had a deal, and took my deal (took donation) and took potential RTL users

I don't want to have anything with common with that guy due to his unethical behaviour. I have never yet experienced such a desperate case of copycating and straight-up stealing of work and of users (he wasn't even been able to find users on his own, just used my work, my SDK and everything to jumpstart his project).

closed 2022-11-13 by openshwprojects

TurkeyMan, 2022-11-13:
Okay, but will you support RTL8710 in your project?

openshwprojects, 2022-11-13:
RTL8710 is 99% done: https://github.com/openshwprojects/OpenRTL8710BN
But I don't have that chip still. If you want to contact me, write to me there: https://www.elektroda.com/rtvforum/topic3866123-1020.html

TurkeyMan, 2022-11-14:
I made posts for some devices, I'll add the TuyaMCU dumps when I create them.

TurkeyMan, 2022-11-14:
If you want to share an ebay (or similar) link to an item close to you, I'll order you one.

openshwprojects, 2022-11-14:
You can do a donation via Paypal, and then I will try to find some device that may be interesting. Also, send me a message on elektroda, because Github does not sadly have private messages, as far as I know.

Still, thanks for contribution! Your support is very valuable.

TurkeyMan, 2022-11-14:
It seems I posted on the wrong topic; your earlier link lead to "Home Automation System" and I just posted there, but I guess it should be moved to "Electronics teardowns", assuming you have rights to move threads?

openshwprojects, 2022-11-14:
Thanks, I will reply to those topics.

Regarding the forum issue, I have moderator rights but I wouldn't worry about that. The main administrator of Elektroda is preparing a little update of forum structure soon and topics will be moved anyway. We are planning to split "Teardowns" section into two sections - IoT teardowns and generic teardowns. This is because, at the current state of things, the "Teardowns" section is for all kind of devices - even VCR players, TVs, etc, you know, all stuff.

We are putting a lot of effort in making IoT DIY accessible for everyday people so a separate section is a must-have.