openshwprojects/OpenBK7231T_App #465: Our devices list has reached 125 devices point! Go go, get to 150! We need more teardowns, and also some help with teardown reviews would help.

openshwprojects, 2022-11-14:
Here is our devices teardowns frontend:
https://openbekeniot.github.io/webapp/devicesList.html
And here is used database:
https://github.com/OpenBekenIOT/webapp/blob/gh-pages/devices.json
Devices list is also accessible from OpenBeken VUE panel

Progress is getting nicely, but we need more help from community in:
1. creating more device teardowns (do you have a device not listed? Make a detailed description and teardown, post to our reliable hosting Elektroda, and add entry to devices json (or let me know, I can do it for you)
2. validating existing teardowns (some of configurations are not written down from teardowns to JSON, if someone has the time please try review each teardown and update devices.json)

If you want to help, just let me know and start making some teardowns. We want to document as many devices as possible, in a detailed way. Not just single pins config and one photo, but a detailed teardown with full information, module markings, device model , etc.

Gamerayers, 2022-12-03:
I'd happildy support the list with 2 of the treatlife bulbs I have (SL10 and SL20).  Reflashing is SUPER easy using the TuyaCloudCutter, BUT, I can't get the RGBCW working since it only using 4 pwns (RGB + Warm White).  Cool white is created by the RGB.  (See https://github.com/openshwprojects/OpenBK7231T_App/issues/211).  If we can fix this, I'd be THRILLED as I have 8+ bulbs like this.

NielsPiersma, 2022-12-24:
Added Kruidvat and Action plugs yesterday :)

closed 2023-01-20 by openshwprojects