openshwprojects/OpenBK7231T_App #466: S9070B Support

ewandank, 2022-11-15:
Hi there, 
absolutely loving this project to flash devices which i thought I would have to throw out as they were not tasmota compatible.
Just cracked open a magic home strip and I am wondering if it is possible to flash? It has a s9070b chipset, as below. 
I struggled to find any documentation as to what the chip is, but I can confirm its not ESP based. 
[image]

leoshusar, 2022-11-15:
The manufacturer is Suzhou SmartChip Semiconductor ([source](https://templates.blakadder.com/unsupported/magichome_ZJ-WFMN-E.html))

openshwprojects, 2022-11-15:
Is there a SDK?

deepcoder, 2022-11-15:
Perhaps something of use at these two:

https://github.com/9lyph/CVE-2020-27199

https://gitter.im/tinkerman-cat/espurna?at=5e8f6cd0cc370f0b07d3d8ec

divadiow, 2025-12-04:
@ewandank did you ever capture boot log from TX? Might be a BL602, which is supported