openshwprojects/OpenBK7231T_App #467: Can't configure pin roles to read energy meter?

TurkeyMan, 2022-11-15:
Firstly, the [pin roles document](https://github.com/openshwprojects/OpenBK7231T_App/wiki/Pin-Roles) on the wiki doesn't appear to be up to date.

I'm trying to configure [this device](https://www.elektroda.com/rtvforum/viewtopic.php?p=20285596#20285596)
The only 2 roles I can see to read inputs are ADC, and dinput. Neither of them seem to do anything useful to read the energy meter on any of the connected input pins.

Also from that device, note P14 which seems to control the power LED. From power-on (while booting up), the LED is half-bright. Once booted, setting to high turns it off, low turns on at full brightness. I can't find a way to control the brightness. PWM/PWM_n doesn't seem to do anything at all; it just stays on. Perhaps this is a DAC pin?

valeklubomir, 2022-11-15:
Energy metering is usually done by special IC (BL0937(pin roles)/BL0942(UART)/CSE7766(UART).....).
So you need disassemble and make some photos.
Control LEDs do not support brighness adjust.

openshwprojects, 2022-11-15:
I am updating only main readme on github repo main page.

btsimonh, 2022-11-15:
@TurkeyMan - feel free to update the wiki..... :).

TurkeyMan, 2022-11-15:
I can't update the pin roles, I have no idea what they do, or what the complete set of roles are... that's why I'm referring to the wiki! :P

TurkeyMan, 2022-11-15:
Okay, I've worked it all out and updated the forum post https://www.elektroda.com/rtvforum/viewtopic.php?p=20285596#20285596

Closing this... the wiki should still be updated though.

closed 2022-11-15 by TurkeyMan