openshwprojects/OpenBK7231T_App #468: Water Valve SM-AW713

Jaguaza, 2022-11-15:
Good morning.

I need help to make this device:

Water Valve SM-AW713

Tuya WiFi Water Valve, Gas shut off controller, Alexa compatible, Google Assistant, Smart Wireless Control, Tuya Smart Life app
https://a.aliexpress.com/_Exjvhr7

[image]

This device contains a WB3S chips and I really have read everything but I don't know what to do sorry.

I can perform a disarming guide with my eyes closed but I don't know how to add this to home Assistant.

openshwprojects, 2022-11-15:
That's a new device I haven't seen before. I will try to help you, but could you make a teardown for our database here:
https://openbekeniot.github.io/webapp/devicesList.html
 so we can help other users?

Regarding the configuration... well, please include more photos in teardown. We need to know which WB3S pins connect where. Which pin is relay, etc, which is button. 

Or is it a TuyaMCU device? With so little photos is hard to tell yet.

valeklubomir, 2022-11-15:
We have button and LED
There should be endpoints for OPEN/CLOSE valve
And probably Motor driver: Forward/Reverse.

RX1/TX1 - seems to be unconnected.
I see 5Pins(possible 6) soldered to PCB.

Jaguaza, 2022-11-15:
> We have button and LED
> 
> There should be endpoints for OPEN/CLOSE valve
> 
> And probably Motor driver: Forward/Reverse.
> 
> 
> 
> RX1/TX1 - seems to be unconnected.
> 
> I see 5Pins(possible 6) soldered to PCB. 
> 
> 

@openshwprojects 

When I get home I'll take a better quality photo.

And from other points. When we have all the information I will compile it and publish it in the database so that buyers can benefit from this help.

When I had Esp I used these pins in Tasmota:
[image]

And this configuration in Esphome:

`binary_sensor:
  - platform: gpio
    pin:
      number: GPIO13
      mode: INPUT_PULLUP
      inverted: True
    name: "Button"
    id: button
    internal: True
    on_press:
      - switch.toggle: relay_template

status_led:
  pin:
    number: GPIO4

light:
  - platform: binary
    id: valve_status_light
    name: "Main Water Valve Status"
    output: valve_gpio5

output:
  - platform: gpio
    id: valve_gpio5
    pin:
      number: GPIO5
      inverted: True

switch:
  - platform: gpio
    pin: GPIO12
    id: relay
  - platform: template
    name: Main Water Valve
    id: relay_template
    icon: 'mdi:pipe'
    lambda: |-
      if (id(relay).state) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
      - light.turn_on: valve_status_light
      - switch.turn_on: relay
    turn_off_action:
      - light.turn_off: valve_status_light
      - switch.turn_off: relay`

Thanks!!!

btsimonh, 2022-11-15:
why are there so many devices I want!!! :).

Jaguaza, 2022-11-15:
> why are there so many devices I want!!! :).

Hahahahahaha.

Who said that with home automation you can save money? This is an excuse we say at home.

Hajahahaha

openshwprojects, 2022-11-15:
"Relay1", so it has a simple Relay? Relays are obviously already supported.

Jaguaza, 2022-11-15:
> "Relay1", so it has a simple Relay? Relays are obviously already supported.

If it is a relay. In an hour and a half, two I'll send you more pictures.

Jaguaza, 2022-11-15:
@openshwprojects , @valeklubomir 

I already have the photos.

The engine has a limit switch, I show in the photos.

The logic of the led works like this:
-When the valve is open the led is on.
-When the valve is closed the led is off.

I hope all this is useful for you.

Pictures:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

valeklubomir, 2022-11-15:
Device does not use bridge driver for controlling motor.
Power Supply is 24VDC. Motor direction control is wired through relay and there are mechanical end switches.
WB3S model just changes polarity of the RELAY and motor starts and runs till end switch stops it by interruptin circuit.
Device works like relay. Writing 0 turn valve into closed position and Writing 1 turn valve into open position. 
LM2596S-3.3 is switching power supply for 3.3V for WB3S module.
I think only 3 PINS have any active role. LED, BUTTON, RELAY. 
I think no extra firmware changes are neccesary. Layout for module WB3S is similar to ESP-12.
This pins are soldered to PCB:
GPIO5   - P8 
GPIO12 - P24 
GPIO13 - P6
GPIO0   - P1 
Other seems not connected. Reset circuit is connected to EN. 
I am not sure which pin are realy used, P1 or P8. GPIO pins are from TAS config. 
But from PCB photo I think used are (P6, P24, P1). It should be measured with multimeter.
All pins could be also checked by PIN monitor. Button can be pressed and change will be visible. 
LED pin will show as 1. because LED active state is 0. Relay will show as 0, to activate relay 1 must be written to Pin.

Jaguaza, 2022-11-16:
> Device does not use bridge driver for controlling motor. Power Supply is 24VDC. Motor direction control is wired through relay and there are mechanical end switches. WB3S model just changes polarity [...]

Tell me how you want me to measure and I will measure continuity, voltage or whatever I need, but forgive my clumsiness, tell me to measure this with this and I will do it.

valeklubomir, 2022-11-16:
Measure continuity between Pins marked between KEY and module pins, LED and module pins. 
[image]
This pin would be controlling RELAY. 
[image]
Pinout for WB3S is here: [WB3S](https://developer.tuya.com/en/docs/iot/wb3s-module-datasheet?id=K9dx20n6hz5n4)

valeklubomir, 2022-11-16:
[image]
99% the yellow pins are used for LED, BUTTON, RELAY. 
Blue mark - I am not sure if the soldering has connection to module. Bad photo.

Jaguaza, 2022-11-16:
> [image] El 99% de los pines amarillos se utilizan para LED, BOTÓN (TECLA), RELÉ. Marca azul: no estoy seguro de si la soldadura tiene conexión con el módulo. Mala foto.
> 
> EDITAR: Mi apuesta: IZQUIERDA: LED y relé Derecha: Botón (TECLA)

How do I measure, do I check voltage when I activate the relay drive by putting the pins on the solder and gnd?

Right side:
[image]
Left side:
[image]

Jaguaza, 2022-11-16:
@valeklubomir , @openshwprojects.

It can be closed, when trying to measure I had to make a short and the chip does not respond, there is no light in the led, device off line in the app and current reaches the chip.  :(

Jaguaza, 2022-11-16:
Well, tomorrow I'll get another one from a friend that I can keep. What do we do?

valeklubomir, 2022-11-16:
Best way to examine connections is to use multimeter and measure continuity with acoustic signaling.
[image]
BEST NOT TO DO IT under voltage. Accidental shortcuts could damage the electronics.

Jaguaza, 2022-11-16:
My friend brought me his, so there's no need to wait until tomorrow.

This is the only one that gives me continuity:

[image]

The LED does not give continuity and does not beep.

valeklubomir, 2022-11-16:
You have detected where is button. 
Assign pin to button funcion. 
Try other pins setup as LED or RELAY and test with button, you should be able to control output state and either LED or RELAY should act.
May be LED is driven through resitor. And you will not have continuity detection, just measured lower resistance than other pins.
May be LED is wired parallel to RELAY signal. And is not controlled by pins.