openshwprojects/OpenBK7231T_App #469: TONGOU TO-Q-SY1-JWT din rail power meter.

valeklubomir, 2022-11-15:
[image]
New device: 
Chipset: CBU
Button, 2X LED, BL0942 energy meter.
[image]
TODO: Relay control. This relay is driven by Driver bridge. Forward direction to close relay, revesre direction to open relay.
Controlled by 2 pins. 
New PinRoles? Driver? How to do it? Any suggestion? I think this would help also #468

openshwprojects, 2022-11-15:
That;s a great finding, please submit teardown and all information according to instruction here:
https://openbekeniot.github.io/webapp/devicesList.html
then you can edit devices.json or I will do it myself:
https://github.com/OpenBekenIOT/webapp/blob/gh-pages/devices.json

Well...  yes, most likely new pin roles and a new drv_SOMETHING.c . I can do that for you, or I can help you to write the driver yourself.

valeklubomir, 2022-11-15:
I just need suggestion. Best would be driver with utilization of QuickRun and pin roles for BridgeForward/BridgeReverse signals.
I have still measure original signals. I will connect osciloscope.

I have not yet disassembled the relay, but I presumme polarized relay. 
[image]
I have searched for this IC and came out MD7620 Mingda. Bi-directional magnetic latching relay driver.

Controlled by 2 pins and original chinese PDF show signals for closing and openning relay.

openshwprojects, 2022-11-15:
A driver file, an entry in drivers array, and maybe some commands for that driver?

Also feel free to create a special pin roles for that driver and a special button role (if you need it).

For example, I created a "Button_NextColor" etc that way.

Of course, we also have a scripting system (and we can bind events to eg. button presses), but my assumption is that it's better to do things automatically because we are making firmware also for users who don't know how to write a script

valeklubomir, 2022-11-15:
Measured the relay characteristics. 
It is polarized relay. 
Forward pulse switches the RELAY ON and reverse polarity switches the RELAY OFF. 
In normal state relay does not need to be powered. 
Pulse length is specified to 100ms. 
-  PinRoles: BridgeForward, BridgeReverse
-  Driver: MotorBridge
-  Command: LinkToChannel, PulseLength

This could also work with Valve device.

valeklubomir, 2022-11-17:
TearDown:
[https://www.elektroda.com/rtvforum/topic3934580.html](https://www.elektroda.com/rtvforum/topic3934580.html)
My first post ever. I think it need some management.
I am not able to edit device.json

openshwprojects, 2022-11-20:
Hey @valeklubomir , sorry, I just forget about this for the last two days.

Are the two SDK PRs tested and ready to merge now?

valeklubomir, 2022-11-21:
I think yes. I have tested it. Ask also @btsimonh for his changes.
But the changes are not so radical. Just some analytics for IRQ/FIQ.
T version I have resolved all the warnings in the compilation.
Maybe LINUX GCC is very sensitive, and may be other build environvements are less sensitive, but I like clean build more.
Warnings should not be ignored.

tomanyusers, 2022-12-02:
I have another device that has the same relay driver. Looks like there are several of the same devices under different brands. 
https://www.elektroda.com/rtvforum/topic3938206.html

I have 4 of these devices so looking forward to a solution to this.

valeklubomir, 2022-12-02:
I will try to do Pull Request. Firmware on my side passed tests, but I have some other issues on progress, so I hope we can soon release it.

keyz182, 2023-01-08:
I have the same device listed in the title. I was able to get this to "work" with `1.15.297` by setting P24 to `Rel`, and P26 to `Rel_n`, with both set to the same channel index. I also set P9 to `LED_n`, and P17 to `Btn`, both with index 0 so that the button and LED works. 

It gets very warm though, with no load connected. I assume this is because the relay is only expecting a pulse, not a constant voltage. 

I was able to get the power monitor bit working by just setting `startDriver BL0942` in the startup command.

openshwprojects, 2023-01-08:
This device is still missing from database. Can anyone submit a detailed teardown?
https://openbekeniot.github.io/webapp/devicesList.html
@valeklubomir  any updates or should I add driver myself?
@keyz182 as a temporary solution, I could try writing a script for you but I am not sure if it makes sense as @valeklubomir  already has a driver?
https://github.com/valeklubomir/OpenBK7231T_App/blob/main/src/driver/drv_bridge_driver.c

tomanyusers, 2023-01-09:
> This device is still missing from database. Can anyone submit a detailed teardown? https://openbekeniot.github.io/webapp/devicesList.html 
Here is the teardown: https://www.elektroda.com/rtvforum/topic3934580.html

keyz182, 2023-01-09:
@openshwprojects I'm in no rush, so no need for scripts. I found this thread after I had figured most of it out myself, so was just documenting what I found. 

I'm happy to wait until @valeklubomir submits the PR.

keyz182, 2023-01-13:
I grabbed the code that @valeklubomir created and tested it out, seems to work.

valeklubomir, 2023-02-11:
@openshwprojects @keyz182 @tomanyusers 
Sorry for my absence late months I had a lot work, was not able to spent much time on this.
I have kept my fork up to date with repository, but now it is 120 commits ahead and I don't know if it is possible to create 
pull request. And I don't wanna discard all my changes and start from scratch.

May be other approach will be better way. ....................

I hate to point out, but very often during merging a lot of warnings pop up and I like to keep the code as clean as possible.

I have progressed with some other platforms like w600/W800. Currently I am strugling with BL602.
My intention is more on target for making new designs, not only to teardowns. 
To use available modules (BK7231N/T, w600, w800, BL602) for DIY devices and more. 
This need much studiyng and experimentation.

And so far I have not figured out How to add any device to database.

valekCN, 2023-02-13:
PR released. closing.

closed 2023-02-13 by valeklubomir