openshwprojects/OpenBK7231T_App #471: BL0937 pins

TurkeyMan, 2022-11-15:
When I change BL0937 pin assignments and save, it doesn't seem to take effect immediately. I notice I need to restart every time.
Is it possible to make these take effect when pressing save without having to reboot the device?
The trouble is, I have a device where the BL0937 is hidden under a board that's soldered in place, so I need to guess the IO pins for CF, CF1, SEL. There's 5 GPIO pins available; quite a large number of possible combinations.

valeklubomir, 2022-11-16:
So far automatic driver restart when pins are changes is not implemented. 
Changing pin configuration has multiple options, what and how to restart. 
Automatic operation would again result in full reboot. 
Since command stopDriver/restartDriver are not implemented, executing manual command startDriver BL0937 will have no effect. Also none of the drivers do not have implemented stop driver action (TODO), reboot is only option at moment.
Question is how beneficial would be action stopDriver/restartDriver could have for user handling? @openshwprojects

TurkeyMan, 2022-11-16:
It's definitely beneficial and sign of mature project, but it's probably low priority.

openshwprojects, 2022-11-17:
@valeklubomir shutdown functions were not written because there was no need to. Right now, I added a quick fix for @TurkeyMan :
https://github.com/openshwprojects/OpenBK7231T_App/commit/49300049588fe02040d81571e4670d61c87cd80c
The build is not released, can you @valeklubomir  maybe check if it's working? Or @TurkeyMan , if you really have BL0937...