openshwprojects/OpenBK7231T_App #473: Trimming BL602 build for ota on the Sonoff B05-BL bulb.

protectivedad, 2022-11-16:
The OpenBL602 image is too big to send OTA to the Sonoff B05-BL-A19 bulb.

First I am not even sure it can be upgraded using the OTA. The Sonoff website says it can but I haven't tried it since I have no firmware small enough that would allow me to do it. After looking at the SDK ansd trying things like nuttx I have concluded that my best bet is to trim down the OpenBL602 build. Since I have found things that I think are redundant I figured it would be helpful if I generated PRs as I did them.

I'm fairly new to the PR process but I'll try. I opened this report just to get feed back and suggestions.

btsimonh, 2022-11-16:
recommend starting with the map file, if the build makes one.
Also, Alpha has had a lot of text cut out.  BUT... it's under suspicion of causing death on N at the moment...  you could safely cherrypick the getcommands commits, I think, which remove descriptive text from source, and put it in .md files.

openshwprojects, 2023-05-24:
@protectivedad  we have merged your changes, thanks. Now Ota file went from 424 KB to 392KB...

closed 2023-05-24 by openshwprojects

xAbdaziz, 2024-02-12:
@protectivedad Hi, I'm sorry for bothering you
Were you able to flash it using DIY mode successfully?
I'm afraid that i might brick mine and i don't have the required tools to flash it using UART.

protectivedad, 2024-02-13:
> @protectivedad Hi, I'm sorry for bothering you Were you able to flash it using DIY mode successfully? I'm afraid that i might brick mine and i don't have the required tools to flash it using UART.

I didn't my gaze was drawn away elsewhere. I did use the great youtube video provided by "Elektroda com" to dismantle the bulbs in a non destructive way and flash via UART. I don't think I have any original bulbs around to test. Sorry.