openshwprojects/OpenBK7231T_App #475: no connection to wifi when flag 18 is set

esnet146, 2022-11-16:
no connection to wifi when flag 18 is set
([LED] Smooth transitions for LED (EXPERIMENTAL)

- Version 1.15.38
- Device RGBCW lamp
- Chip/model: BK7231T
- Config: P7-SM2135DAT,P8-SM2135CLK, set flag 4,10,12

Steps to reproduce the behavior:
1. Set Flag 18
2. Press Submit
3. Power off
4. Power on
No device online, no wifi router log entries. 

If the lamp was turned on before the power was turned off, it will turn on with a delay.
transitions working

openshwprojects, 2022-11-17:
This is a very useful report. It seems that doing a SM2135 write every 5 ms (quick tick) is too much for WiFi.

I will have to find a better write rate... or... hmm..  SM2135 is currently using rtos_delay_milliseconds(SM2135_DELAY); .... what if we replace it with... with a nop sleep.

Can you check that binary? Of course, if you're willing to take the risk..
[OpenBK7231T_App_1.0.0.zip](https://github.com/openshwprojects/OpenBK7231T_App/files/10034371/OpenBK7231T_App_1.0.0.zip)

valeklubomir, 2022-11-17:
Question, why is not I2C used instead of software driven IIC communication?

esnet146, 2022-11-18:
> Just power cycle it once to have it connect to wifi again.

Waited. did not connect

> Can you check that binary? Of course, if you're willing to take the risk..

No problem. In the evening

openshwprojects, 2022-11-18:
@valeklubomir because LED drivers are not "real I2C", they have some slight change in I2C protocol and futhermore they can be on any pins, not just on hardware I2C pins.

In the mentioned binary, I changed  rtos_delay_milliseconds(SM2135_DELAY); to sleep with nops instead of rtos and we will see what happens.

I think that we will also soon need to change the time of quick tick from 5 ms to maybe 20ms or more (what is the highest acceptable value?), because @btsimonh  did some progress with power saving/sleeping and supposedly it's possible to save power by sleeping for a, let's say, 15 ms between quick ticks;

esnet146, 2022-11-18:
> OpenBK7231T_App_1.0.0.zip
Great. Exactly what I wanted!

esnet146, 2022-11-18:
sec. color is real. sm2135.

closed 2022-11-18 by esnet146