openshwprojects/OpenBK7231T_App #476: Death on N: WARN: TCPIP mutex is locked

btsimonh, 2022-11-17:
**Describe the bug**
My N device (IR recv/send + wifi LED) dies regularly.
@openshwprojects and I suspected Alpha, but I don't think there is anything in there which affects much,

Yesterday, it died after 1000s, 3000s, then this morning, ~40000s.
It's still has DHCP lease, but no ping, so I hooked up serial log without restarting the device, and see:

```
Info:MAIN:Time 48343, idle 360/s, free 81960, MQTT 0(681), bWifi 1, secondsWithNWARN: TCPIP mutex is locked
oWARN: TCPIP mutex is locked
WARN: TCPIP mutex is locked
WARN: TCPIP mutex is locked
WARN: TCPIP mutex is locked
Ping -1, socks 2/38
Info:MAINWARN: TCPIP mutex is locked
:Time 48344, idle 0/s, free 81960, MQTT 0(681), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 48345, idle 456/s, free 81960, MQTT 0(681), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 48346, idle 242/s, free 81960, MQTT 0(681), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 48347, idle 28/s, free 81960, MQTT 0(681), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 48348, idle 73/s, free 81960, MQTT 0(681), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 48349, idle 36/s, free 81960, MQTT 0(681), bWifi 1, secondsWithNoPing -1, socks 2/38
WARN: TCPIP mutex is locked
Info:MAIN:Time 48350, idle 127431/s, free 81960, MQTT 0(681), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:GEN:dhcp=0 ip=192.168.1.60 gate=192.168.1.1 mask=255.255.255.0 mac=a0:92:08:88:b6:6e
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-63,ssid=MYSSID,bssid=00:37:b7:30:e0:0e ,channel=11,cipher_type:
WARN: TCPIP mutex is locked
Info:MAIN:Time 48351, idle 184545/s, free 81960, MQTT 0(681), bWifi 1, secondsWithNoPing -1, socks 2/38
```

Interesting facts:
The last weblog had a seconds count of 
`Info:MAIN:Time 48025, idle 178327/s, free 81832, MQTT 1(667), bWifi 1, secondsWithNoPing -1, socks 2/38`

yet one of the first serial logs I saw had:
`Info:MAIN:Time 47967, idle 179474/s, free 81832, MQTT 1(667), bWifi 1, secondsWithNoPing -1, socks 2/38`

so, it must have restarted at some point in the night.. yet by my calculations 48000s is 13 hours, and I flashed it about that amount of time ago - strange to see the last weblog be from the future?.

@valeklubomir - any ideas about TCP mutex locking not being released in N SDK?

**Firmware:**
- Version (main + startup mod comits from alpha)
- Device? N
- Chip/model: BK7231N
- Device config: no SSDP.  MQTT running, IRRx + IRTx + Wifi LED.

**To Reproduce**
Run N on Alpha (.35.alpha.2?)

btsimonh, 2022-11-17:
more info:

the message is there from boot on serial logs:

First occurrence:
```
add is_broadcast_ether_addr
sta:255, vif:0, key:1
add hw key idx:1
ctrl_port_hdl:1
me_set_ps_disable:840 0 0 0 0 3
sta_ip_start

configuring interface mlan (with DHCP client)WARN: TCPIP mutex is locked
WARN: TCPIP mutex is locked
Info:MAIN:wl_status 10
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTED
ip_addr: 3c01a8c0
new dtim period:2

new ie: 0 : 4d 59 53 53 49 44
new ie: 1 : 82 84 8b 96 c 12 18 24
new ie: 3 : b
new ie: 30 : 1 0 0 f ac 4 1 0 0 f ac 4 1 0 0 f ac 2 c 0
new ie: 2d : ee 11 1b ff ff 0 0 0 0 0 0 0 0 0 0 1 0 0 0 0 0 0 0 0 0 0

Info:MAIN:wl_status 11
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTED
```

btsimonh, 2022-11-17:
Further notes.

The message is NORMAL.  The place it comes from should check that the core IS locked, and the messages is printed if it IS locked.
It DOES NOT warn us if the core is NOT locked, or the function not called from the LWIP core thread.

Serial log which covers web logs topping at 679s:
```
WARN: TCPIP mutex is locked
Info:MAIN:Time 676, idle 186586/s, free 70176, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 3/38
WARN: TCPIP mutex is locked
Info:MAIN:Time 677, idle 182327/s, free 81872, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
WARN: TCPIP mutex is locked
Info:MAIN:Time 678, idle 182156/s, free 81872, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
WARN: TCPIP mutex is locked
Info:MAIN:Time 679, idle 173430/s, free 81872, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
WARN: TCPIP mutex is locked
Info:MAIN:Time 680, idle 181367/s, free 81872, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:GEN:dhcp=0 ip=192.168.1.60 gate=192.168.1.1 mask=255.255.255.0 mac=a0:92:08:88:b6:6e
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-57,ssid=MYSSID,bssid=00:37:b7:30:e0:0e ,channel=11,cipher_type:
WARN: TCPIP mutex is locked
Info:MAIN:Time 681, idle 182214/s, free 70176, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 3/38
WARN: TCPIP mutex is locked
Info:MAIN:Time 682, idle 189523/s, free 70176, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 3/38
WARN: TCPIP mutex is locked
Info:MAIN:Time 683, idle 187610/s, free 70176, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 3/38
WARN: TCPIP mutex is locked
Info:MAIN:Time 684, idle 189735/s, free 70176, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 3/38
WARN: TCPIP mutex is locked
Info:MAIN:Time 685, idle 190272/s, free 70176, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 3/38
```

btsimonh, 2022-11-17:
More info:
ALL mqtt_ calls must be made from tcp_thread.  (even to check if still connected)...
There seems some way to book a callback from tcp_thread so that you can achieve this.

Investigating.

valeklubomir, 2022-11-17:
If logging is stopped, then device somehow got locked in endless loop. Without possibility to switch other task.
From what I heard that IR is pretty consuming on CPU time, can you not try to disable or put some more logs into IR?
Warn TCPIP mutex is locked means that TCP-IP thread is using mutex and possible collision could cause system locked.
But you always have main Info message after lock. LWIP does not offer solution for this check, I placed WARN instead of ASSERT which would cause immediate crash. I have examined it and when function enter Mutex it wait till it is released and the continues. Critical would be that somehow TCPIP would run into state when lock is active and it tries to lock it second time this would immediately create endless loop, because same thread can not wait on unlock which must be done in same thread.

btsimonh, 2022-11-17:
the serial log continues...  but the TCP layer is trashed in some way.

The warning is telling us that a function is being called where the mutex is expected to be taken, but is not.
For us, this includes asking if the mqtt_client is still connected - that is the regular one (I instrumented the warning with the caller address!!! :) ).

I strongly suspect that some of our calls to mqtt are then trashing something inside the tcp_thread's domain.  But I think it should not be too hard to make all mqtt calls from within the tcp_thread through the use of the callback mechanism.  Will investigate more tonight.

Of course the N tcp crash I experience may be completely unrelated :(.

openshwprojects, 2022-11-19:
MQTT is currently off (but it was on yesterday for 4 hours or so) but my N device is still standing with build main from 2 days ago, see screenshot:
[image]
so maybe the culprit is in Alpha?

openshwprojects, 2022-11-27:
It's fixed now I think? After @btsimonh  changes?
[image]

closed 2022-11-27 by openshwprojects