openshwprojects/OpenBK7231T_App #477: Device group intermittently not working

ferbulous, 2022-11-17:
**Describe the bug**
On first boot, device group would function as expected, however after a few minutes or so, it's no longer sending/receiving device group commands. On two occassion when device group isn't running there's no dgr related logs in the console and another 'DGR no socket'.
Restarting the device would fix this, however the same issue happens later.

**Firmware:**
- Version : 1.15.36
- Device: LED bulb
- Chip/model: BK7231N
- Device config - Smartrul SMR-DP005-WIF-A19

**To Reproduce**
Steps to reproduce the behavior:
1. Enable device group, all option enabled (power, color, brightness)
2. Add the same group to other openbk device/tasmota.
3. Power toggle the tasmota/change brightness syncs with openbk on smartrul led bulb
4. After a few minutes, openbk doesn't seem to respond.
5. An hour later, device group resumes working and stops again, which seems to happen more random.

**Logs**
If applicable, add screenshots to help explain your problem.
```
`Info:DGR:no sockets, will retry creation soon, in 17 secs
Info:MAIN:Time 70825, idle 468928/s, free 78072, MQTT 1(5), bWifi 1, secondsWithNoPing -1, socks 25/38
Info:DGR:no sockets, will retry creation soon, in 16 secs
Info:MAIN:Time 70826, idle 447182/s, free 78072, MQTT 1(5), bWifi 1, secondsWithNoPing -1, socks 25/38
Info:DGR:no sockets, will retry creation soon, in 15 secs
Info:MAIN:Time 70827, idle 440403/s, free 78072, MQTT 1(5), bWifi 1, secondsWithNoPing -1, socks 25/38
Info:DGR:no sockets, will retry creation soon, in 14 secs
Info:MAIN:Time 70828, idle 459156/s, free 78072, MQTT 1(5), bWifi 1, secondsWithNoPing -1, socks 25/38
Info:DGR:no sockets, will retry creation soon, in 13 secs
Info:MAIN:Time 70829, idle 462216/s, free 78072, MQTT 1(5), bWifi 1, secondsWithNoPing -1, socks 25/38
Info:DGR:no sockets, will retry creation soon, in 12 secs
Info:MAIN:Time 70830, idle 922792/s, free 78072, MQTT 1(5), bWifi 1, secondsWithNoPing -1, socks 25/38
Info:GEN:dhcp=0 ip=192.168.1.134 gate=192.168.1.1 mask=255.255.255.0 mac=c8:47:8c:00:00:00 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-66,ssid=Suliana,bssid=a0:ab:1b:84:8c:d0 ,channel=11,cipher_type:
Info:DGR:no sockets, will retry creation soon, in 11 secs
Info:MAIN:Time 70831, idle 449083/s, free 78072, MQTT 1(5), bWifi 1, secondsWithNoPing -1, socks 25/38
Info:DGR:no sockets, will retry creation soon, in 10 secs
Info:MAIN:Time 70832, idle 461820/s, free 78072, MQTT 1(5), bWifi 1, secondsWithNoPing -1, socks 25/38`
```
**Additional context**
Add any other context about the problem here.

openshwprojects, 2022-11-18:
"Info:DGR:no sockets, will retry creation soon, in 16 secs" 
This means that there is no DGR socket and DGR driver will start in 16 seconds (if possible).

Maybe I just need to make the delay smaller for you?

ferbulous, 2022-11-19:
DGR driver is loaded, but there's no dgr related logs even when I'm sending dgr commands 

```
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-44,ssid=Suliana,bssid=f8:e9:03:c9:29:b9 ,channel=10,cipher_type:
Info:MAIN:Time 441, idle 459075/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 442, idle 464304/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 443, idle 462996/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 444, idle 446098/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 445, idle 450697/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 446, idle 380547/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 447, idle 421909/s, free 70240, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 5/38
Info:MAIN:Time 448, idle 453768/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 449, idle 462075/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 450, idle 461815/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:GEN:dhcp=0 ip=192.168.1.247 gate=192.168.1.1 mask=255.255.255.0 mac=38:1f:8d:05:72:e2 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-43,ssid=Suliana,bssid=f8:e9:03:c9:29:b9 ,channel=10,cipher_type:
Info:MAIN:Time 451, idle 461056/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 452, idle 452984/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 453, idle 463116/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 454, idle 465547/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 455, idle 462123/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 456, idle 456356/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 457, idle 461135/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 458, idle 454426/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 459, idle 462499/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 460, idle 451736/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:GEN:dhcp=0 ip=192.168.1.247 gate=192.168.1.1 mask=255.255.255.0 mac=38:1f:8d:05:72:e2 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-43,ssid=Suliana,bssid=f8:e9:03:c9:29:b9 ,channel=10,cipher_type:
Info:MAIN:Time 461, idle 453719/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 462, idle 448913/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 463, idle 450422/s, free 73328, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 5/38
Info:MAIN:Time 464, idle 442281/s, free 73544, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 5/38
Info:MAIN:Time 465, idle 451016/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 466, idle 902612/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 467, idle 448017/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 468, idle 460256/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 469, idle 453717/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 470, idle 455556/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:GEN:dhcp=0 ip=192.168.1.247 gate=192.168.1.1 mask=255.255.255.0 mac=38:1f:8d:05:72:e2 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-42,ssid=Suliana,bssid=f8:e9:03:c9:29:b9 ,channel=10,cipher_type:
Info:MAIN:Time 471, idle 445936/s, free 70240, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 5/38
Info:MAIN:Time 472, idle 460729/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 473, idle 457850/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 474, idle 454907/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 475, idle 445862/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 476, idle 453382/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 477, idle 456869/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 478, idle 462600/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 479, idle 444844/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 480, idle 461510/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:GEN:dhcp=0 ip=192.168.1.247 gate=192.168.1.1 mask=255.255.255.0 mac=38:1f:8d:05:72:e2 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-43,ssid=Suliana,bssid=f8:e9:03:c9:29:b9 ,channel=10,cipher_type:
Info:MAIN:Time 481, idle 455411/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 482, idle 456280/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 483, idle 435285/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 484, idle 458944/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
[... cut ...]
Info:MAIN:Time 515, idle 459823/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 516, idle 456732/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 517, idle 455478/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Drv DGR is already loaded.
Info:MAIN:Time 518, idle 441428/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
Info:MAIN:Time 519, idle 456093/s, free 81936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 4/38
```

ferbulous, 2022-12-05:
Tested 1.15.128 (N & T)
still getting no sockets errors after device online after a few hours
```

Info:DGR:no sockets, will retry creation soon, in 0 secs
Info:DGR:DRV_DGR_CreateSocket_Send: failed to do socket
Info:MAIN:Time 45649, idle 190365/s, free 82720, MQTT 1(9), bWifi 1, secondsWithNoPing -1, socks 25/38
Info:DGR:no sockets, will retry creation soon, in 4 secs
Info:MAIN:Time 45650, idle 200139/s, free 82720, MQTT 1(9), bWifi 1, secondsWithNoPing -1, socks 25/38
Info:GEN:dhcp=0 ip=192.168.1.237 gate=192.168.1.1 mask=255.255.255.0 mac=84:e3:42:bf:f1:e7 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-57,ssid=Suliana,bssid=a0:ab:1b:84:8c:d0 ,channel=6,cipher_type:CCMP
Info:DGR:no sockets, will retry creation soon, in 3 secs
Info:MAIN:Time 45651, idle 197359/s, free 82504, MQTT 1(9), bWifi 1, secondsWithNoPing -1, socks 26/38
```

```
Info:DGR:DRV_DGR_CreateSocket_Send: failed to do socket
Info:MAIN:Time 46224, idle 454138/s, free 71440, MQTT 1(218), bWifi 1, secondsWithNoPing -1, socks 25/38
Info:DGR:no sockets, will retry creation soon, in 4 secs
Info:MAIN:Time 46225, idle 453717/s, free 71440, MQTT 1(218), bWifi 1, secondsWithNoPing -1, socks 25/38
Info:DGR:no sockets, will retry creation soon, in 3 secs
Info:MAIN:Time 46226, idle 453422/s, free 71440, MQTT 1(218), bWifi 1, secondsWithNoPing -1, socks 25/38
Info:DGR:no sockets, will retry creation soon, in 2 secs
Info:MAIN:Time 46227, idle 445228/s, free 71440, MQTT 1(218), bWifi 1, secondsWithNoPing -1, socks 25/38
Info:DGR:no sockets, will retry creation soon, in 1 secs
Info:MAIN:Time 46228, idle 453911/s, free 71440, MQTT 1(218), bWifi 1, secondsWithNoPing -1, socks 25/38
Info:DGR:no sockets, will retry creation soon, in 0 secs
Info:DGR:DRV_DGR_CreateSocket_Send: failed to do socket
Info:MAIN:Time 46229, idle 454390/s, free 71440, MQTT 1(218), bWifi 1, secondsWithNoPing -1, socks 25/38
```