openshwprojects/OpenBK7231T_App #479: Device group - MQTT crash for devices not configured as light bulbs

ferbulous, 2022-11-17:
**Describe the bug**
MQTT payload sent to openbk device not configured as light bulbs (device tested is a wall switch) crashes openbk.
However for devices that's configured as light bulb openbk works as normal

**Firmware:**
- Version: 1.15.36
- Device - Wall Switch
- Chip/model: CB3S
- Device config : Tuya LSPS5

**To Reproduce**
Steps to reproduce the behavior:
1. Device group enabled. All options are selected.
2. Toggle on/off from homeasssistant dashboard or send payload from mqtt explorer
4. Openbk becomes unresponsive.
5. Power cycle is required to reboot the device again

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

openshwprojects, 2022-11-17:
Thanks, I think we are aware about that right now.

This is most likely the same issue as with bulbs - one cannot do UDP send from MQTT callback. The correct approach is to make a packets queue in DGR driver and send them on quick.
I pushed UNFINISHED, commented out version for reference:
https://github.com/openshwprojects/OpenBK7231T_App/commit/6ee700933189df8d9740d3c29cb1cd5d5d3e2e1b

I can try updating it FASTER for you, maybe today or tomorrow, please stand by and help us with testing when it's needed.

Currently I am working on Windows port (so users can try writing scripts on windows and learn faster) and on self-test system, but I will try to find some time.

openshwprojects, 2022-11-18:
Yup, indeed it crashed for me. I will look into it now, if time allows...

ferbulous, 2022-11-18:
> Can you check? [22dac4b](https://github.com/openshwprojects/OpenBK7231T_App/commit/22dac4bf582be1e385f6ce6b9fe1e98cad0d6a8f)

Thanks, this update has fixed the crash issue

ferbulous, 2022-11-19:
I'm not sure it just crashed or disconnect wifi after an hour, will continue testing and verify

openshwprojects, 2022-11-21:
@ferbulous  is it still happening?

btsimonh, 2022-11-21:
ref incoming MQTT, we currently do evil inside the mqtt_callback call.  This is resolved in mqtt_test, but is still in test as I have some unreliability on N.  Latest news is that the unreliability may come from the IR interrupt as @valeklubomir hinted at.

openshwprojects, 2022-11-21:
I have moved DGR sending from MQTT callback to QuickTick (Added a send out packet queue), but idk.
@ferbulous  you will need to test new mqtt_test2 soon

but @ferbulous , you are not using IR interrupt, are you?

ferbulous, 2022-11-21:
@openshwprojects no, not using IR interrupt
Just the wall switch device and no crashes so far

openshwprojects, 2022-11-29:
@ferbulous  how it's going with DGR? Any futher input?

ferbulous, 2022-11-30:
@openshwprojects this crash issue can be closed now.
As for the DGR, still getting ‘DGR no sockets’ in the logs’.
Will continue testing with latest firmware