openshwprojects/OpenBK7231T_App #481: BK7231T devices not connecting to network due to corrupted flash data

rogere66, 2022-11-17:
I've had 2 LED bulbs failing to connect to WiFi recently. One was a new install while the second stopped connecting after a power cycle. I managed to connect in AP mode (using 5 power cycles), but was still not able to make them connect to the WiFi network. At last I managed to connect them to a network with different SSID and Password.

Some further testing revealed that the problem was related to a flash memory block at address 0x1E2000: deleting the block fixed the problem - restoring the old data broke it again. After digging trough the code I found that the block contains configuration data used by the network code for "Fast Connect". One particular byte defines the WiFi security type, and this byte had the wrong value on the failing devices. The original Tuya code (V2.9.15) also use this memory block, but for some reason leave wrong data in the block. OpenBK stores the block correctly, but at least at one occasion the block was corrupted during a power cycle.

So, do we need "Fast Connect"? Probably not - it will only save 3-4 seconds. And it is already disabled for BK7231N because of address conflict.
I suggest we disable it also for BK7231T (comment out RL_SUPPORT_FAST_CONNECT in role_launch.h).

openshwprojects, 2022-11-18:
First of all, are you sure it's not the "blank RF section - empty MAC" issue? And that "Restore RF partition" from Web App won't solve your problem?

Still, I have a slight feeling that I experienced this issue once. It was reported on Elektroda. User managed to get it working by clearing whole 2MB of device flash. But strangely enough, it was for N, unless I'm much mistaken, while N already had it disabled...

It might be possible that Tuya updated Fast Connect partition format and we are using old SDK, so they are not compatible?

I also remember disabling fast connect on N platform and not seeing it on T, but maybe I am wrong.. yes "it is already disabled on BK7231N" you are correct. This is because it was overwriting our config structure. 

I see you also proposed a fix... that's great, @btsimonh , what do you think about it? I would like to hear your opinion, but most likely, we will just disable it as @rogere66  said.

rogere66, 2022-11-18:
I was not able to run Web App when in AP mode, but restored the RF block (TLV) trough serial port - same problem.
To be properly implemented Fast Connect really need fallback to normal connect in case it fails.

openshwprojects, 2022-11-18:
@rogere66 you are correct, but this is the issue related to Beken/Tuya SDK and not to our code. Their code is very buggy.

I've been thinking about the other solution - what about clearing the fast connect block from code when connection fails several times?

btsimonh, 2022-11-18:
I think before clearing it as a default approach, we should document all it's contents, and understand them.
It's a nested TLV form, and the code around it gets quite complex.  Originally we tried to use it to store some of our data,  but soon abandoned that through it being a little complex, and wanting more control over OUR flash storage.
I've seen reports of less-good wifi after clearing - I would not like to lose information which may be a factory calibration for a specific device...

rogere66, 2022-11-18:
The TLV block is quite different. The fast connect block is simple and well defined by struct RL_BSSID_INFO_T. Clearing it will force a normal connect and a the block will be restored with updated values. Either method will fix the problem - clearing the block or disabling fast connect.