openshwprojects/OpenBK7231T_App #482: wifi channel select menu

euro2, 2022-11-18:
I have multiple AP's with the same SSID. But on different channels. It just wont connect to 1 particular router but will on others in the house. I think chipset of router or whatever. But Since its closed to it will always try that one. 

I would like to see an menu where i can choose which SSID+CHANNEL to connect to.

leoshusar, 2022-11-19:
I'd just like to point that it's usually done by selecting BSSID (the MAC address of the AP you want to connect to), not an SSID + channel.
Channels can easily be changed, for example UniFi has a function named "Nightly Channel Optimization" that you can turn on and it will periodically scan and change channels for the least interference. I'm not saying this is a good function, just that locking to channel is not the best idea.

euro2, 2022-11-24:
Im fine with selecting the BSSID, as long as i can pinpoint a router.