openshwprojects/OpenBK7231T_App #484: FR: Sending "backlog" with MQTT and custom MQTT-messages possible?

Handy-E, 2022-11-20:
Can anybody tell me how to use "backlog" with MQTT?
E.g., sending "cmnd/dev1/startScript" with payload "myScript" works. But how to send a payload like "backlog led_dimmer 10; led hue 200"? "cmnd/dev1/cmnd" is wrong ;-) Hints, please!
My Problem is, that some automation systems like openhab send all data at once, like "270,96,100" (hue,sat,bri). I am able to transform data there, so instead a payload like "backlog led_hue %s; led_saturation %s; ..." is send. "%s" is replaced with corresponding values. But sending 3 MQTT-messages seems not possible without a lot of scripting, so "backlog" could be a saver here.

Also if openbeken itself would be able to send hue,sat,bri,... at once would be a really cool feature.
Maybe, a MQTT-customizer or payload-customizer or something? One could define a "myCustomLEDgetter" like "$led_hue, $led_dimmer, ..." and afterwards you get MQTT-messages "dev1/myCustomLEDgetter/get" with payloads "270, 100, ...".
Proposal for the x-mas whishlist??? :-)

openshwprojects, 2022-11-20:
I think you might be approaching it from the wrong angle.
What kind of system do you need to support?
What kind of standard is used in OpenHab? 

If the is some kind of standard needed for OpenHab, I can add support for that directly. Just tell me specs. It will be possible to enable it through flags.

Handy-E, 2022-11-20:
THANKS @openshwprojects !
Maybe, your right. I just thought a "MQTT-customizer" would be an universal tool one could use for almost every system at the end.

openHAB uses "Bindings". With them it supports multiple devices\systems. So, HomeAssistant and Homie are some possible standards. But there are others, too:
[openHAB MQTT Bindings](https://www.openhab.org/addons/bindings/mqtt/)

Even generic MQTT support (which I use most) is possible:
[MQTT Things and Channels](https://www.openhab.org/addons/bindings/mqtt.generic/)
This is the most flexible one, because you can get most different systems under one hat. They just need to "speak" MQTT.

Handy-E, 2022-12-01:
At least somebody who can tell me the MQTT-topic for sending "backlog"-commands to the device?
How to send a payload like "backlog led_dimmer 10; led hue 200"? What's the right topic ("cmnd/<device>/???") ?

NielsPiersma, 2022-12-24:
I can confirm that you can inject configuration via http. Not tested with mqtt (yet).

I case you want to provision, you can use the following;

http://[ip]/cm?cmnd=backlog+SSID1%2wifi-net;Password1%20wifi-pwd;MqttUser%20user;MqttPassword%pwd;MqttHost%20ip-homeassistant

or the one line I use for my smartplugs

http://x.x.x.x/cm?cmnd=backlog+SetPinRole%206%20LED_n;SetPinChannel%206,1;SetPinRole%207%20Btn;SetPinChannel%207,1;SetPinRole%208%20Rel;SetPinChannel%208,1;SetPinRole%2026%20wifiled;SetPinChannel%2026,0;SSID1%20WIFINET;Password1%20WIFIPWD;MqttUser%20xyz;MqttPassword%20secretpwd;MqttHost%20y.y.y.y;restart

Niels

Handy-E, 2022-12-25:
> http://x.x.x.x/cm?cmnd=backlog+SetPinRole%206%20LED_n;SetPinChannel%206,1;SetPinRole%207%20Btn;SetPinChannel%207,1;SetPinRole%208%20Rel;SetPinChannel%208,1;SetPinRole%2026%20wifiled;SetPinChannel%2026,0;SSID1%20WIFINET;Password1%20WIFIPWD;MqttUser%20xyz;MqttPassword%20secretpwd;MqttHost%20y.y.y.y;restart
Yes, I know I can send via HTTP like `http://x.x.x.x/cm?cmnd=backlog ...` . But my problem is that I need to do the backlog-stuff with MQTT. So far I had no luck sending backlog-payloads to devices because I don't know the correct syntax to do so.
I think my used MQTT-topic for that is just wrong. I tried with `cmnd/dev1/cmnd` and `cmnd/dev1/cm`, both with payload `backlog led_dimmer 10; led hue 200` but nothing happens. Or probably "backlog" is just not possible with MQTT.

NielsPiersma, 2022-12-26:
Did you try with the + after cmnd and %20% for a space? I'll do some research later this week. Openbeken is still bèta and not all functions are fully implemented and operational.

Niels

http://x.x.x.x/cm?cmnd=backlog+SetPinRole%206%20LED_n;SetPinChannel%206,1;SetPinRole%207%20Btn;SetPinChannel%207,1;SetPinRole%208%20Rel;SetPinChannel%208,1;SetPinRole%2026%20wifiled;SetPinChannel%2026,0;SSID1%20WIFINET;Password1%20WIFIPWD;MqttUser%20xyz;MqttPassword%20secretpwd;MqttHost%20y.y.y.y;restart

NielsPiersma, 2022-12-26:
I've been trying today, both other than power 1 and power 0 I cannot get any command except from MQTT.

So;

topic:
   obkBF512F1E/1/set
with payload:
   0
or
   1
Is working.
But any other commands are not.

Or I am doing something wrong.

I tested pushing "Publishall" as a payload, but nothing happens. "Publishall" does work when executed from the GUI, but not as an MQTT topic command.

NIels

NielsPiersma, 2022-12-26:
I took a look at the source (https://github.com/openshwprojects/OpenBK7231T_App/blob/main/src/mqtt/new_mqtt.c)
https://github.com/openshwprojects/OpenBK7231T_App/blob/682b7a281e2a2a95fc1c95be5b7f8e10c404431d/src/mqtt/new_mqtt.c#L569
And that leads me to believe I should be able to use cmnds (as with tasmota)
 "// this accepts cmnd/<clientId>/<xxx> to receive data to set channels"

I'll play around and see if this is a correct assumption.

NIels

NielsPiersma, 2022-12-26:
@Handy-E, Well been playing around, and to be honest, I can only publish on/off to the device. I don't seem to be able to publish any other configuration parameters.

I use the MQTT cmnd's quite often. For instance, for enabling and disabling Syslog for groups of MQTT devices (mostly Tamota boxes). Or for mass updating my wifi password in a controlled matter.

Maybe opening a ticket with the request of currently available MQTT cmnds would be possible.

I am not a programmer, so coding is not my piece of cake. But testing and diagnosing are the things I love to do.

Niels

openshwprojects, 2022-12-26:
Huh? cmnd topic should accept any command. Even led_basecolor_rgb is a command. Let me check...

openshwprojects, 2022-12-26:
Fix part 1 : https://github.com/openshwprojects/OpenBK7231T_App/commit/681f6e904392ae60dd52463c0d8b0d57da4ce4d4 @btsimonh  can help with part 2, I am not sure if I should change his MQTT buffers code, right now up 2048 bytes command arg

NielsPiersma, 2022-12-26:
Thanks,

Will test tomorrow. I think it is for me the lack of finding the documentation as a non programmer.
Your examples show me why it was not working for me
Niels

Handy-E, 2022-12-26:
Hey. I never had problems sending regular commands with MQTT, e.g. `cmnd/dev1/startScript` with payload `myScript` or `cmnd/dev1/led_basecolor_rgb` with payload `506070` always worked.
I just couldn't figure out the topic for "backlog". But now, thanks to the examples from @openshwprojects, I know that the right syntax is `cmnd/dev1/backlog`. This means that "backlog" can be interpreted like a command, too. That wasn't clear to me! I always tried to send "backlog" WITHIN the payload. That was my fault.

I never had problems with command length. I think 2048 bytes is enough, no need to change that!

openshwprojects, 2022-12-26:
Please check if everything works now ok and close issue if there are no further problems. Thank you for testing

openshwprojects, 2022-12-26:
btw I think wwe need commands to write autoexec remotely (LFS write command?) in the future, for configuring devices remotely, @NielsPiersma