openshwprojects/OpenBK7231T_App #485: Setting flag smooth transitions for LED makes MQTT unusable

Cybermb, 2022-11-21:
**Describe the bug**
I have recently modified nedis smartlife RGBCW led strip controller with this firmware. With some configuration and lots of doc reading everything works fine. I have integrated controller with home assistant using MQTT. I love the feature flag to smooth color transition, except it only works when changing colors in controllers webui. It does not work with MQTT commands or trough webapp status page. Instead, it changes the color for a fraction of the second and then returns to the value set in controllers webui

**Firmware:**
- Version [e.g. 1.15.55]
- Chip/model: BK7231N

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "configure general"
2. Set flag [LED] Smooth transitions for LED (EXPERIMENTAL)
3. Send MQTT command or in webapp > status > change values
4. Led strip blinks for a fraction of second and returns to the value set by controller.

openshwprojects, 2022-11-22:
I think it may  have been just fixed tomorrow or today with the MQTT update, but take care with flashing, it's an initial version of a more stable MQTT approach.

Cybermb, 2022-11-23:
@openshwprojects do you mean version 1.15.70? I have flashed it but issue still remains

GravityRZ, 2022-11-23:
was there not an issue with timing with the sm2135 loosing network connection when combined with the smooth transitions option. i thought they fixed that but this looks related

Cybermb, 2022-11-29:
Is it planned to be fixed? @openshwprojects