openshwprojects/OpenBK7231T_App #492: [W600] OpenBK stops responding after a random length of time (minutes to hours)

CaptClaude, 2022-11-22:
I successfully flashed a W600-Pico dev board with 1.15.36 and after some tinkering, determined the Pin-to-GPIO mapping.
After some more tinkering,  I got it talking to my MQTT broker and can turn pins on and off with Node-Red.

The board will simply stop working after a while. Sometimes a along while:
`Info:MAIN:Time 87126, idle 0/s, free 30296, MQTT 0(5445), bWifi 1, secondsWithNoPing 1, socks 2/8
`
Sometimes a short while:
`Info:MAIN:Time 6746, idle 0/s, free 27920, MQTT 0(3), bWifi 1, secondsWithNoPing 5, socks 3/8`

When it stops, of course it does not respond to a ping and I need to push the reset button go get it back.
The long example was before I had the right IP address for my broker but the second was publishing just fine. In fact, I have recorded the **connected** topic and it will go from `online` to `offline` and back to `online` and eventually `offline` for good.

I now have the log set to ExtraDebug but it's not giving me any clues that I can see as to why it simply stops.  I also have the USB port talking to a serial monitor and I'm seeing the same info as in the WebApp.

Any suggestions on how to determine why it's doing this and how to make it stop?

Also: what are the standard topics that OpenBK publishes?  How often is uptime published?

A view of the online/offline behavior:
[image]

btsimonh, 2022-11-22:
Hi @CaptClaude 
could you try the latest release?
But bear in mind this has not been tested on W600, so feedback appreciated.

The mqtt has been modified to not process received commands inside the callback, but instead defer to 'QuickTick'.
This (along with other Beken platform related changes) improved the MQTT on N and T chips...

best regards,

Simon

CaptClaude, 2022-11-25:
Flashed the newest release a few hours ago. I have a Node-Red instance running and flip-flopping two GPIO pins (ON-OFF ... OFF-ON ... every 5 seconds). Also have a serial terminal monitoring the serial logging and the webapp running watching the same thing. So I'm keeping it busy, Not exactly a torture test, but better than nothing.
Sample output from the webapp logs window:
```
idle 0/s, free 27920, MQTT 1(17), bWifi 1, secondsWithNoPing 1, socks 3/8
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic w600/0/set
Info:MQTT:MQTT in topic w600/0/set
Info:MQTT:channelSet part topic 0/set
Info:MQTT:channelSet channel 0
Info:MQTT:MQTT client in mqtt_incoming_data_cb data is 1 for ch 0
Info:GEN:CHANNEL_Set channel 0 has changed to 1 (flags 0)

Info:MAIN:Channel has changed! Publishing change 0 with 1 
Info:MQTT:Publishing val 1 to w600/0/get retain=0
Info:MAIN:Time 15103, idle 0/s, free 30296, MQTT 1(17), bWifi 1, secondsWithNoPing 1, socks 2/8
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic w600/1/set
Info:MQTT:MQTT in topic w600/1/set
Info:MQTT:channelSet part topic 1/set
Info:MQTT:channelSet channel 1
Info:MQTT:MQTT client in mqtt_incoming_data_cb data is 0 for ch 1
Info:GEN:CHANNEL_Set channel 1 has changed to 0 (flags 0)

Info:MAIN:Channel has changed! Publishing change 1 with 0 
Info:MQTT:Publishing val 0 to w600/1/get retain=0
Info:MAIN:Time 15104, idle 0/s, free 30296, MQTT 1(17), bWifi 1, secondsWithNoPing 1, socks 2/8
Info:MAIN:Time 15105, idle 0/s, free 30296, MQTT 1(17), bWifi 1, secondsWithNoPing 1, socks 2/8
Info:MAIN:Time 15106, idle 0/s, free 30296, MQTT 1(17), bWifi 1, secondsWithNoPing 1, socks 2/8
Info:MAIN:Time 15107, idle 0/s, free 30296, MQTT 1(17), bWifi 1, secondsWithNoPing 1, socks 2/8
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic w600/0/set
Info:MQTT:MQTT in topic w600/0/set
Info:MQTT:channelSet part topic 0/set
Info:MQTT:channelSet channel 0
Info:MQTT:MQTT client in mqtt_incoming_data_cb data is 0 for ch 0
Info:GEN:CHANNEL_Set channel 0 has changed to 0 (flags 0)

Info:MAIN:Channel has changed! Publishing change 0 with 0 
Info:MQTT:Publishing val 0 to w600/0/get retain=0
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic w600/1/set
Info:MQTT:MQTT in topic w600/1/set
Info:MQTT:channelSet part topic 1/set
Info:MQTT:channelSet channel 1
Info:MQTT:MQTT client in mqtt_incoming_data_cb data is 1 for ch 1
Info:GEN:CHANNEL_Set channel 1 has changed to 1 (flags 0)

Info:MAIN:Channel has changed! Publishing change 1 with 1 
Info:MQTT:Publishing val 1 to w600/1/get retain=0
```
I will let it run and see what happens overnight.

It will need to run without failing for 24hr before I consider it good enough to use.  Nevertheless, it is working.  I need to write something to check to see if it's online and if not, flag it.
I think you can consider this release preliminarily "tested" on a W600. If there are any specific tests you want me to try, I will if I can.
Good work!

openshwprojects, 2022-11-25:
Thanks, i am eager to hear the results

CaptClaude, 2022-11-25:
Well, we're not out of the woods yet. It ran for about 5 hours and then quit responding. It no longer responds to a ping.
The last few seconds from the serial monitor are below.  Suggestions?
```
2022-11-25 02:17:09 --> Info:MQTT:channelSet part topic 1/set
2022-11-25 02:17:09 --> 
2022-11-25 02:17:09 --> Info:MQTT:channelSet channel 1
2022-11-25 02:17:09 --> 
2022-11-25 02:17:09 --> Info:MQTT:MQTT client in mqtt_incoming_data_cb data is 1 for ch 1
2022-11-25 02:17:09 --> 
2022-11-25 02:17:09 --> Info:GEN:CHANNEL_Set channel 1 has changed to 1 (flags 0)
2022-11-25 02:17:09 --> 
2022-11-25 02:17:09 --> 
2022-11-25 02:17:09 --> Info:MAIN:Channel has changed! Publishing change 1 with 1 
2022-11-25 02:17:09 --> 
2022-11-25 02:17:09 --> Info:MQTT:Publishing val 1 to w600/1/get retain=0
2022-11-25 02:17:09 --> 
2022-11-25 02:17:09 --> Info:MAIN:Time 34229, idle 0/s, free 30296, MQTT 1(39), bWifi 1, secondsWithNoPing 1, socks 2/8
2022-11-25 02:17:09 --> 
2022-11-25 02:17:10 --> sta:rssi=35,ssid=EnchantedApril Ü. .Ü. ,bssid=200185a8:0b:cd:2c:ca:41 ,channel=134673643,cipher_type:WPA2_PSK_CCMP
2022-11-25 02:17:10 --> 
2022-11-25 02:17:10 --> Info:MAIN:Time 34230, idle 0/s, free 30296, MQTT 1(39), bWifi 1, secondsWithNoPing 1, socks 2/8
2022-11-25 02:17:10 --> 
2022-11-25 02:17:10 --> Info:GEN:ip=192.168.1.104,gate=255.255.255.0,mask=192.168.1.254,dns=192.168.1.254
2022-11-25 02:17:10 --> 
2022-11-25 02:17:11 --> Info:MAIN:Time 34231, idle 0/s, free 27920, MQTT 1(39), bWifi 1, secondsWithNoPing 1, socks 3/8
2022-11-25 02:17:11 --> 
2022-11-25 02:17:12 --> Info:MAIN:Time 34232, idle 0/s, free 30296, MQTT 1(39), bWifi 1, secondsWithNoPing 1, socks 2/8
2022-11-25 02:17:12 --> 
2022-11-25 02:17:13 --> Info:MAIN:Time 34233, idle 0/s, free 30080, MQTT 1(39), bWifi 1, secondsWithNoPing 1, socks 3/8
2022-11-25 02:17:13 --> 
2022-11-25 02:17:13 --> Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic w600/0/set
2022-11-25 02:17:13 --> 
2022-11-25 02:17:13 --> Info:MQTT:MQTT in topic w600/0/set
2022-11-25 02:17:13 --> 
2022-11-25 02:17:13 --> Info:MQTT:channelSet part topic 0/set
2022-11-25 02:17:13 --> 
2022-11-25 02:17:13 --> Info:MQTT:channelSet channel 0
2022-11-25 02:17:13 --> 
2022-11-25 02:17:13 --> Info:MQTT:MQTT client in mqtt_incoming_data_cb data is 1 for ch 0
2022-11-25 02:17:13 --> 
2022-11-25 02:17:13 --> Info:GEN:CHANNEL_Set channel 0 has changed to 1 (flags 0)
2022-11-25 02:17:13 --> 
2022-11-25 02:17:13 --> 
2022-11-25 02:17:13 --> Info:MAIN:Channel has changed! Publishing change 0 with 1 
2022-11-25 02:17:13 --> 
2022-11-25 02:17:13 --> Info:MQTT:Publishing val 1 to w600/0/get retain=0
2022-11-25 02:17:13 --> 
2022-11-25 02:17:14 --> Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic w600/1/set
2022-11-25 02:17:14 --> 
2022-11-25 02:17:14 --> Info:MQTT:MQTT in topic w600/1/set
2022-11-25 02:17:14 --> 
2022-11-25 02:17:14 --> Info:MQTT:channelSet part topic 1/set
2022-11-25 02:17:14 --> 
2022-11-25 02:17:14 --> Info:MQTT:channelSet channel 1
2022-11-25 02:17:14 --> 
2022-11-25 02:17:14 --> Info:MQTT:MQTT client in mqtt_incoming_data_cb data is 0 for ch 1
2022-11-25 02:17:14 --> 
2022-11-25 02:17:14 --> Info:GEN:CHANNEL_Set channel 1 has changed to 0 (flags 0)
2022-11-25 02:17:14 --> 
2022-11-25 02:17:14 --> 
2022-11-25 02:17:14 --> Info:MAIN:Channel has changed! Publishing change 1 with 0 
2022-11-25 02:17:14 --> 
2022-11-25 02:17:14 --> Info:MQTT:Publishing val 0 to w600/1/get retain=0
2022-11-25 02:17:14 --> 
2022-11-25 02:17:14 --> Info:MAIN:Time 34234, idle 0/s, free 30296, MQTT 1(39), bWifi 1, secondsWithNoPing 1, socks 2/8
2022-11-25 02:17:14 --> 
2022-11-25 02:17:15 --> Info:MAIN:Time 34235, idle 0/s, free 27920, MQTT 1(39), bWifi 1, secondsWithNoPing 1, socks 3/8
2022-11-25 02:17:15 --> 
2022-11-25 02:17:16 --> Info:MAIN:Time 34236, idle 0/s, free 27920, MQTT 1(39), bWifi 1, secondsWithNoPing 1, socks 3/8
2022-11-25 02:17:16 --> 
2022-11-25 02:17:17 --> Info:MAIN:Time 34237, idle 0/s, free 25544, MQTT 1(39), bWifi 1, secondsWithNoPing 1, socks 4/8
2022-11-25 02:17:17 --> 
2022-11-25 02:17:18 --> Info:MAIN:Time 34238, idle 0/s, free 23168, MQTT 1(39), bWifi 1, secondsWithNoPing 1, socks 5/8
2022-11-25 02:17:18 --> 
```

btsimonh, 2022-11-25:
did the serial stop?  i.e. is it a crash, or just loss of connectivity?
If serial still running, worth looking at your router to see if it lost actual wifi connection.  On N/T, I did have the situation that it lost IP but was still connected to the router.  Bumping it from the router forced a re-connect, and re-established IP.  Not a soluion, but more info reference the issue...

CaptClaude, 2022-11-26:
The serial stopped and there was no response to a ping. I pushed the reset button and it's up again (has been for a while) but without MQTT on/off of GPIO pins. I will leave it this way and see if the MQTT is an issue.
I am still seeing online/offline being reported:
[image]
Uptime is now >30k.
Freeheap is down from >30k to 23k or so. Dropped to 18296 at one point.  General trend is down.  
A while ago I dealt with sporadic crashes of an ESP8266 that was eventually traced to a memory leak which slowly decreased the freeheap until the processor crashed. That was with the Arduino ESP library and once they found/fixed the error, there were no more crashes.  I applaud you for tracking freeheap in MQTT.

Freeheap trend:
[image]

CaptClaude, 2022-11-26:
Crashed again. Last reported uptime = 43273, freeheap = 25544. Serial stopped as well.  No response to ping.
Might need to build in some more diagnostic information. I'm happy to test whatever you create.

openshwprojects, 2022-11-27:
@CaptClaude before we fix the issue, would adding a simple watchdog or just a simple heap monitor mechanism that restarts the device when heap goes below certain point, help with your use case?

CaptClaude, 2022-11-29:
Would a heap-dependent resetting watchdog help?  I suppose, but it's not really a good fix.  That said, I don't want to request a fix unless it's something _**everyone**_ needs.  I am playing with OpenBK on the W600 because I have several and because it supports MQTT which I don't think I can get with Micropython. 
I think you should put this on your roadmap because I can write a simple wifi-connected program in Micropython that runs and runs and never stops.  There is something in in the W600 version of OpenBK that is causing a problem with stability.  I will do some more experimenting and see if it has anything to do with how I configured my device. Judging from the output, I don't think it's the heap shrinking to the point of a crash.  I've seen that in the ESP8266 and you could watch the heap shrink down to a couple of kb before the device crashed.  The heap here does not seem to be doing that.  

If you have some debugging suggestions, I'm listening. I'll let you know whatever I find out.

Thanks for asking.

CaptClaude, 2022-12-02:
Trying OpenW600_1.15.123.fls using one of the W600 device templates as a starting point (outlet with LED, Relay, WiFi LED and one additional LED.monitoring the Log and serial output while toggling the LEDs via MQTT every 5s.

Would someone kindly point me to a description of the MQTT topic structure? What internal information is published? Might be helpful for tracking health online.

CaptClaude, 2022-12-02:
Drops wifi and doesn't rejoin, requires reset to reconnect.  Stays running as I can see from the serial port. Investigating further.

btsimonh, 2022-12-02:
@CaptClaude -
I had a similar issue on beken N where it apparently drops wifi, but continues to run.  Feels like a DHCP issue...
In my case, I have OpenWRT on my router, and the device is still connected to wifi, but not responding on IP.
Kicking it off the router at the wifi level forces it to reconnect....
Would be interesting to know if this is also the case for the W600.... (hence my previous question about if serial continued.......).
If you cant kick it off the wifi via a router UI, you could try a router reboot?

CaptClaude, 2022-12-02:
@btsimonh 
I will need to look into that but rebooting the router to test this would be very disruptive.  I can seeing about setting it up on a different subnet where I have some more control.

CaptClaude, 2022-12-05:
Additional data point: I am using CoolTerm for Mac to monitor the serial output.  After > 12hr, it stopped again.  In CoolTerm, I clicked "disconnect" and then "connect" and I started getting data again.  Next time this happens, I'll check to see if it is responding to ping before disconnect/connect.  After that, I'll stop monitoring via serial and see if it stays connected.
The app log window had likewise stopped and took several seconds to catch up with the serial terminal.