openshwprojects/OpenBK7231T_App #495: MQTT topic update and publish interval configuration

MrDix, 2022-11-22:
Describe the bug**
### MQTT topic related documentation
I was used from tasmota that I can configure the the MQTT _Topic_ and _Full Topic_ with placeholders. so I tried to figure out where to configure the topic related names in OpenBK7231T_App firmware.
If I reed the [documentation](https://github.com/openshwprojects/OpenBK7231T_App#mqtt-topics-of-published-variables) the topic base is 'OBK_DEV_NAME' which I interpreted as a placeholder variable somewhere. 
The [command documantation](https://github.com/openshwprojects/OpenBK7231T_App/wiki/Console-Commands) informs at the _pubish command_ section that the base name would be always be 'obk0696FB33' - which makes no sense as we typically have more than one device connected to the same MQTT server and we need to have individual topic base names.
### MQTT topic related configuration
After being unable to find any topic related configuration I thought that maybe 'OBK_DEV_NAME' means that the firmware uses the configurable _ShortName_ or _Full Name_ as a base topic. So I set different strings into these fields and checked the MQTT messages. My test device has gotten the base topic _Aubess-Power-Monitor-Switch-1_ which used to be my configured _Full Name_ of the device in the past.
I've changed the _Full Name_ since these days several times but the base topic of the MQTT messages never got updated. I tried different simple names, rebooted the device, kept it unplugged for hours - but nothing helped to change the topic base.
Now I do not know if the base topic can only be set once and can never be changed or if this is a bug and changing the device full name used to work in the past and is broken in newer versions of the firmware?
Currently I'm running firmware version 1.15.70 and I pretty sure changing the base topic by changing the full device name also did not work in my previously installed version 1.15.30. I'm not sure about versions before.
### MQTT topic related target
What I actually wanted to configure is for example a base topic of ' OpenBK7231T_App/Aubess-Power-Monitor-Switch/groundfloor/livingroom/uplight-1'. Each topic level can be parsed into a variable in Telegraf. So I can configure a Telegraf parser for all 'OpenBK7231T_App/Aubess-Power-Monitor-Switch' devices as they all produce the same MQTT message structure and therefor can be parsed the same way. I can tag the collected power readings with the the firmware type 'OpenBK7231T_App', the device model 'Aubess-Power-Monitor-Switch', the floor level 'groundfloor', the room name 'livingroom', the location local device name 'uplight-1' and write those tags as metadata along with the actual power readings into my InfluxDB time series DB. This allows me to filter measurements as needed for my visualization and reports or ask the home automation system via location filtered voice commands for values. At least this is the way I do it with my tasmota based devices.
### MQTT topic related update intervals
The [command documantation](https://github.com/openshwprojects/OpenBK7231T_App/wiki/Console-Commands) also contains this side information:
> Some MQTT variables are being published only at the startup, some are published periodically (if you enable "broadcast every minute" flag), some are published only when a given value is changed. Below is the table of used publish topics (TODO: add full descriptions)

Unfortunately I can not find the _broadcast every minute_ flag in the flasgs section of the _Configure General_ menu. Where can I set this this flag or read about other flags like this one?

My device is:

**Firmware:**
- Version: 1.15.70
- Device: [Aubess - Power Monitor Switch](https://www.elektroda.com/rtvforum/topic3887748.html#20266899)
- Chip/model: BK7231N
- Device config: 
- - Startup command	startDriver BL0942
| PIN  |  Role      | Channel |
| 8	   |LED_n       | 1       |
| 23	 |Btn         | 1       |
| 24	 |TglChanOnTgl|	1       |
| 26	 |Rel         |	1       |

**To Reproduce**
This is a mix of documentation related and under standing related issues which I tried to point out in bug description already.

**Screenshots**
[image]
[image]

**Additional context**
Thank you very much for all the time and efforts!

openshwprojects, 2022-11-23:
Hello, before I address the main part of your post, I will note this - here is the flag, it's marked as "don't use" but it should be safe now:
[image]

btsimonh, 2022-11-23:
Hi @MrDix
 - if you find the time, please update the wiki in areas where it is incorrect.
There is also now a 'docs/' folder which contains AUTOEXTRACTED commands documentation.
This will improve as people enhance the comments above the commands in the source code and re-run 'npm run getcommands' (you need nodejs installed to run....).

Also worth while adding pages in the wiki for usage scenarios - e.g. describing what you were trying to achieve, and how you achieved it - as guidance for others.

br,
Simon

schneemaier, 2022-11-25:
For me setting the client ID under MQTT settings changes the topic for the device.

+1 to be able to change the broadcast time from 1 minute

Also coming from Tasmota it would be nice to have a similar 'status' MQTT message

openshwprojects, 2022-12-27:
@schneemaier  @MrDix 
Added:
https://github.com/openshwprojects/OpenBK7231T_App/commit/7b1e1ca30d932771ae2aa2d2c8bd678ad0f92064

schneemaier, 2023-01-02:
@openshwprojects Thank you!

I flashed one of my devices with the latest firmware. Am i reading correct that i need to execute command like `backlog mqtt_broadcastInterval 15;` at every startup as the interval value is not stored?

Also not sure what mqtt_broadcastItemsPerSec is good for. Is that to broadcast more than once per every second? Example: `backlog mqtt_broadcastInterval 15; mqtt_broadcastItemsPerSec;` would result in 2 broadcasts every 15 seconds?

Thanks