openshwprojects/OpenBK7231T_App #496: Lightning Semiconductor LN8825 support

4timepi, 2022-11-23:
Hi,
Is there any future plans for supporting Semiconductor LN8825 chip used in Tuya devices? 
https://developer.tuya.com/en/docs/iot-device-dev/hardware?id=Kaiuyfgqxi1cl

I have Intempo LED Strip (EE6715NPFOB) with LN8825B chip. Pictures of PCB in attachments.

[image]
[image]

IceFlinger, 2023-12-24:
I also have this device and can't find any information or datasheets about it anywhere, my closest guess is this TUYA module but I haven't managed to get anything out of it over UART yet. https://developer.tuya.com/en/docs/iot/wifie3lpinmodule?id=K9605uj1ar87n

openshwprojects, 2024-03-29:
Related topic: 
https://www.elektroda.com/rtvforum/topic4023264.html