openshwprojects/OpenBK7231T_App #497: [To devs] Precise timing/bingbang issue on BK7231N

openshwprojects, 2022-11-24:
I am trying to make a driver for basically a WS2812b clone (SM16703P) which requires precise timing on IO output:
[image]
I have already written a WS2812B driver for PIC several years ago and I wanted to take the same approach, just delay enough with NOPs and send bit by bit. Good enough for a restart, we could think of something better later.

The problem is, BK7231N behaves in a way that I cannot predict. 
Here is the working demo version:
https://github.com/openshwprojects/OpenBK7231T_App/commit/7bc04d07d1d535400a76e6745d37c29076f643df
Results:
[image]

This is predictable, but when I just change the number of nops:
[image]
I get that:
[image]

Why am I getting such unpredictable results?

Futhermore, I managed to simplify the current code because the reg write values seem constant for high and low:

#define reg_val_HIGH 0x02
#define reg_val_LOW 0x00
and setting pins boils down to just writing that byte at given address, but it still doesn't fix my issue.

Optimized version:
https://github.com/openshwprojects/OpenBK7231T_App/commit/ddf03290fc90b25c1af481813ccf1d94469d3ff3
It still gives me:
[image]
Version above, but when I remove all nop delays:
[image]
then I get that:
[image]
I have no idea how I get that large seemingly random delays. It didn't happen for me when I was doing WS* driver on PIC, neither when I was generating PAL on PIC18F.

@btsimonh  @valeklubomir  @iprak  ideas?

valeklubomir, 2022-11-24:
I have seen similar behaviour from ESP32. Delay was caused by reading code from flash and writing into cache.
ESP32 uses SPI flash for code. And I think BK7231 is build very similar. Especially that FLASH can be written via SPI.

Work-around was to place the code in instruction RAM, I am not sure if this can be done in BK7231. If code execution from RAM is allowed or can be used. ARM9 is different from LX6.

Work-arround here could be SPI but only with DMA. Prepare whole data stream in RAM and use DMA transfer.

valeklubomir, 2022-11-24:
If you want to do it in code. You should avoid using call to function, only inline code.

openshwprojects, 2022-11-24:
I do not have enough knowledge to do it in the DMA yet. I assumed that I am choosing an easier option.

The code for the last photo basically boils down to:
```

#define REG_WRITE(addr, _data) 	(*((volatile UINT32 *)(addr)) = (_data))

#define SM16703P_SET_HIGH REG_WRITE(gpio_cfg_addr, reg_val_HIGH);
#define SM16703P_SET_LOW REG_WRITE(gpio_cfg_addr, reg_val_LOW);

#define SM16703P_SEND_T0 SM16703P_SET_HIGH; ; SM16703P_SET_LOW; ;

	SM16703P_SEND_T0;
	SM16703P_SEND_T0;
	SM16703P_SEND_T0;
	SM16703P_SEND_T0;
	SM16703P_SEND_T0;
	SM16703P_SEND_T0;
	SM16703P_SEND_T0; 
	SM16703P_SEND_T0;

```

So there is no function calls (if we are talking about the same thing).

The next thing I wanted to try would be putting all 8 port "ticks" in the __asm block, but I do not have enough assembly knowledge to do that, but if that's related to reading the binary code from flash, it also won't help.

I considered taking such approach:
https://github.com/bigjosh/SimpleNeoPixelDemo/blob/master/SimpleNeopixelDemo/SimpleNeopixelDemo.ino
Line 58.

valeklubomir, 2022-11-24:
Check MAP file if function is  linked in RAM.

openshwprojects, 2022-11-24:
Sorry, I didn't see your comment, but I tried modifying the bk7231n_ota.ld and doing a full rebuild and now device dies when executing test command, most likely when doing that jump to the function in RAM. So at least something happens.  It turns out the bk7231_ota.ld is using for building RBL, at least. I will try without the "long_call" attr, also maybe will try to make the function smaller or something.

valeklubomir, 2022-11-24:
Use ram sectio .itcm there are already some function placed. Especially Flash write functions.

openshwprojects, 2022-11-24:
Ok I will try. Ah, I see, that's a good find with that itcm. I can see that some functions are straight up mapped there, vTask stuff etc. That could work.

Moment ago I also tried that approach:
[image]
[image]
but it crashes:
[image]
Will try itcm now.

openshwprojects, 2022-11-24:
Yes! It seems that the ictm approach with directly specyfiyng function name here works:
[image]
[image]
That's perfect! I hope the remaining part of implementation will go without any further issues. Thank you @valeklubomir for pointers, it would take me a great deal of time to find it without your help. I wasn't doing any ESP stuff before Beken, mostly juc PIC18 and they are way simpler.

valeklubomir, 2022-11-24:
BEWARE that .itcm section has limits and during LWIP upgrade it overflowed and there was no warning. 
So please verify MAP file before uploading to BK7231.

openshwprojects, 2022-11-24:
@valeklubomir it seems that I just wanted to write it.  **I can see there is some kind of instability now.** Luckily the device in question is very accessible and has breaked out UART.

valeklubomir, 2022-11-24:
Also you need to make the code uniform. No loop, no jump, no function call. 
Prepare data to RAM and shift direct for array. 

You work with BK7231N?
I am trying it for BK7231T, but linker script does not contain .itcm

openshwprojects, 2022-11-24:
It's on BK7231N, as user requested.
Currently it seems device dies even if I don't start the driver in question, it dies at around 80 second uptime. I am testing more now and also trying to determine where that 4608 here comes from:
[image]

valeklubomir, 2022-11-24:
If you increase LENGTH, you need to decrease ORIGIN
You have to shrink TCM section and reduce RAM reserved by LWIP.
Best to shrink MEM_SIZE in file
platforms/bk7231n/bk7231n_os/beken378/func/lwip_intf/lwip-2.1.3/port/lwipopts.h

Check mapfile and region around 0x003FFFFF - 0x00400100 there could be overlapping section.
Area between 0x00400000 and 0x00400100 is not mentioned in linker script but is is used by INTC
And addressed directly.

openshwprojects, 2022-11-24:
I have wrongly assumed that 0x00400000 to 0x00400100 is free. Ok, I will try now as you said.

openshwprojects, 2022-11-25:
I am back at it. I will try again, also with the memory section resizing approach.

@valeklubomir , do you know why just putting a function there (without changing partitions sizes):
[image]
[image]
leads to generic device instability?

BTW, small hint: if your device is unstable and you want to get it running long enough to do OTA, close all browser tabs with the device except the OTA tab. Even HTTP page while open does refresh calls and they stress device.

openshwprojects, 2022-11-25:
@valeklubomir  it seems I am stuck. I unable to get the rid of instability.

Steps to reproduce:
Option A: just add the mentioned function here:
[image]
Option B:
[image]
[image]
Here is generated map fragment:
[image]

In both cases, after a full rebuild, device networking dies after about 50 to 100 seconds.
The device itself does not die, because I used script: "addRepeatingEvent 2 -1 POWER0 toggle" and the device is still clicking.

The device does not die if I don't place "flash_function" in RAM, even with the other changes kept in place.
The device dies even if I don't start the driver and  even if I don't call "flash_function"

I double checked offsets but I can't see anything wrong. Can you spot some kind of issue?

valeklubomir, 2022-11-25:
I have to harvest BK7231N module from some device and solder wires to log and programming. 
Try new attempt. 
I think there must be still an overlapped section in RAM (TCM/ITCM/BSS), because most of the LWIP variable reside in TCM section.

openshwprojects, 2022-11-25:
I will try again tomorrow, but what do you mean by "try new attempt"? Is there something specific you have in mind?

openshwprojects, 2022-11-25:
@valeklubomir  this is not related to this issue, but when you next time add your changes to OBK, can you please make separate pull requests for each feature? I am asking because last time Github was lagging so badly with many code changes that it was hard to review it. Thanks.

valeklubomir, 2022-11-25:
@openshwprojects What do you mean ? I always comment my changes, at least try to make it understandable.
"try new attempt" 
- I will not change SM16703P driver, I will put test function into test led driver
- You probably have not commited any changes, I do it from scratch, more like clone any existing RAM function.
- Start with smallest implementation and increase with each test

btsimonh, 2022-11-25:
silly thought.  but 'normal' RAM seems to start at 0x400000.  So what is the bit before it?
This may be some special memory mapped stuff, or special SRAM?
Most likely shared ram for network processor...
If trying to run something new from RAM, try changing the ram start from 0x400100, and inserting a section there.
e.g. create a new named section marked rwx starting 0x400100 for the length you need, and then start ram section after that?
But...  also, I'm not sure there is a cache benefit.   It still takes a different number of cycles to lift from normal RAM to cache that to execute directly from cache.  So you may need to get REALLY detailed on ARM, and force the function into cache before executing, in order to get consistent performance.
s
update - this link explains a lot:
https://blog.feabhas.com/2020/11/introduction-to-the-arm-cortex-m7-cache-part-3-optimising-software-to-use-cache/

Seems you would need to use itcm.  But you probably can't make it bigger....
You may need to disable interrupts, backup the content (in case something else has loaded code there and expects it to still be there), load and execute your function, then restore the content before re-enabling interrupts.  Sounds a big faff to me....
May be easier to work out how DMA can be used...

openshwprojects, 2022-11-26:
@valeklubomir Sorry, I didn't mean that your changes are not commented, I am saying that Github page seem to have problems with loading large pull requests on my PC.  No matter who does a larger pull request, I have issues with reading diff...
It happens for me for every github project, not just OBK, also work projects etc etc. Their page is just laggy with larger ccode.

@btsimonh interrupts are already disabled of course, I started with that.

Does all that mean that the ictm section is already 99% full and we will not able to add new function there?

I am wondering if bk can execute function from anywhere in RAM or is that RAM sector somehow special in hardware..

**Guys, I will not be able to test new things for a day or two, I am testing all my available devices stability with new router.**
[image]
[image]
Build on Nov 25 2022 10:36:03 version 1.15.77 seems to be stable 100% well for my usage, but I didn't start stress tests yet. I will test with stress tests later, for now, a basic test.
About 30 devices in network (ping list includes only OBK and Tasmota)

valeklubomir, 2022-11-26:
ARM968E-S has memory areas ITCM (0-4MB), DTCM (4-8MB). AHB (8MB - 4GB)
I - instruction, D - Data. 
ITCM can contain executable code and data
DTCM is data area only, no executable code.
AHB - peripherals, ......

I Presume following memory map:
0x00000000 - 0x001FFFFF - FLASH (2MB)
0x003C0000 - 0x003FFFFF - RAM  (256kB) Mirror A (ITCM)
0x00400000 - 0x0043FFFF - RAM  (256kB) Mirror B (DTCM)

So we ca move ram sections between  0x003C0000 - 0x0043FFFF 
Moving 256kB window within 512kB area.

valeklubomir, 2022-11-26:
This may be also may be reason why issues occur with usage of heap. 
because physically RAM area 003F0000 - 003FFFFF is probably mirrored to 00430000 - 0043FFFF and heap is usuall placed at end of RAM section 0400100 - 042FFFF. And when something writes over address 0042FFFF it actually starts overwriting LWIP variables and is corrupting TCPIP stack and network is dead.

openshwprojects, 2022-11-28:
@valeklubomir  this could explain some things. 

Would it be possible to reserve more memory in that RAM, for all bit bang drivers, also WS2182 b etc? 

SECOND IDEA: That's something I thought about recently. Maybe it would be possible to create one single, let's say, 200 bytes block in that ITCM and just copy the function we want in RAM when it's needed?
Copy that function to specific address... and call it.... and then "forget it"...
This would have an advantage, because we might need to have multiple bit-bang drivers (WS2182B, SM16703, etc etc) but only one active at the time.

So I would imagine something like that:
1. at build time reserve 200bytes in itcm
2. at driver_start for WS28128 /sm16307/ etc etc copy the select function there
3. at driver run call that function

valeklubomir, 2022-11-28:
@openshwprojects 

Creatiing self-modify code, need to suppress several GCC directives. 

- Function should be linear. Only short/relative jumps, no calls to external functions. 
- Best to write it in ASM. Template could be build in flash and copied to ITCM ram section.

But this type of protocol is shared by WS2182B, SM16703, etc. So universal code should be sufficient.

iprak, 2022-11-28:
Christmas time :-) Would it be possible to use/adjust FastLED or NeoPixel library. FastLED does mention support for arm processor so should be close to our chips. 

I just finished experimenting with a modded firmware on ESP to run some strips.

iprak, 2022-11-28:
I only have W600 chip accessible and I am going to try FastLED with it.

valeklubomir, 2022-11-28:
In ESP32 RMT can be used and whole transmission is executed via hardware module.

iprak, 2022-11-28:
> For ESP8266 I suggest SPI based protocol.

From my reading, SPI can only be used if LED strip has separate data and clock connections. I happen to be using WS2812B which is the more common one and has one data connection. FastLED uses clockless approach for controlling those LEDs.

Someone asked a question about W600 and FastLED so it seems that using it with W600 would be worth investigating - https://github.com/FastLED/FastLED/issues/1178

openshwprojects, 2022-12-21:
@valeklubomir any progress on ramfunc?

alcomys, 2023-01-18:
Is there any news on this ramfunc problem?
I did looked at it myself also, but this material goes far beyond my knowledge about programming.
so i hope this can be resolved.
my nice ledstrip laying around in the closet now :(

alcomys, 2023-03-10:
Nobody respond anymore, so i think to sell the ledstrip and buy another type because this is strange.

alcomys, 2023-05-21:
I found this on github.
maybe this can help us.
its about the same timing issue but they found something out.

https://github.com/qmsk/esp/pull/42
best regards

oldies, 2023-07-23:
Hello everyone.
I've bought an "JSK-P22" Wall Light. The control unit is Based on an Tuya CBU Module (BK7321N Chip). The RGB-Led-Pixel (some Kind of SM16703P/WS2812/...) is connected to the MOSI-Pin of the  BK7321N - so it looks like the original Software also use the SPI to drive this Pixel.

So I've taken the example implementations of the spidma and put it together to drive the LED Pixels. The first try seems to work very well. You can find this "proof-of-concept"-implentation here: https://github.com/openshwprojects/OpenBK7231T_App/compare/main...oldies:OpenBK7231T_App:main

To test this you can follow these steps:
1. Start driver
startDriver SM16703P

2. Init Driver
SM16703P_Init <<pixelcount>>

3. Set Pixel
SM16703P_SetPixel 1 255 0 0
SM16703P_SetPixel 2 0 255 0
SM16703P_SetPixel 3 0 0 255

4. Start Output (each call will trigger one
SM16703P_Start

Maybe these could help to develope an universal LED-Pixel-Driver.

Best Regards!

openshwprojects, 2023-07-23:
Very good! Have you considered making a pull request? I don't have SM16703 at hand. How hard would be to adjust it for standard WS?

openshwprojects, 2023-07-26:
I am trying to merge it now. It seems only BK7231N is supported, but that's not a problem for a start.