openshwprojects/OpenBK7231T_App #498: lightbulb does not send out led_dimmer when switched on/off

GravityRZ, 2022-11-25:
v 1.15.70
i noticed that the light only sends out led_dimmer through MQTT when you change the dim
however when you switch on the light there is no way to tell at what percentage the dimrate is.

when i enable flag 2 it sends it out on a regular base but it seems logical we need all the status info when turned on and with every change

openshwprojects, 2022-11-25:
Do you have that "broadcast full light state" or whatever it was called flag enabled?

GravityRZ, 2022-11-25:
i think that is flag 0 and yes it is enabled
this only sends out dimmer and color on change.

strange that nobody else complained about this problem(or i am doing something wrong.

from the gui it works because of flag 17
on=the light goes on
change brightness=the light goes on to the right brightness

from a MQTT perspective
ON=Led_enable is send out and the light goes on at the last brightness setting but because  led_dimmer is not send out the receiving party does not know what brightness to display and sticks at 1%

GravityRZ, 2022-11-28:
any news on this
is it possible i can try this functionality to see if this solves a problem with my setup?

openshwprojects, 2022-11-28:
Wait wait wait, so basically you need... you want OBK to broadcast "led_dimmer" when "led_enableAll" is set to 1?

GravityRZ, 2022-11-29:
yes because some  systems do turn on when the light is switched but if they do not receive the led_dimmer variable that status stays at 1%

maybee create a testversion to see if it fixes my problem

openshwprojects, 2022-11-29:
To be honest, I can't see any drawbacks... ah, may it be, we will see how it works on release.
https://github.com/openshwprojects/OpenBK7231T_App/commit/5b89deb036493109780be05d6997142dbfb3e7fc

GravityRZ, 2022-11-30:
i tried it and it works
in my situation the light still first jumps to 1% and then jumps to the real brightnesss(thanks for this option)

still it shows this light implemtation under domoticz is not working like it should

if it does no harm you can leave it in there but it might be better to remove for now